google-deepmind/mujoco #431: how to fix object sliding problem

vaesong, 2022-08-08:
Hi,

I'm a  student  and I'm trying to use Mujoco for reinforcement learning

I'm having some trouble simulating the xml object file that the object keep sliding on the table.
Is there a problem with my xml file? 

[image]

```XML
<mujoco model="FrontCamera">
    <asset>
      <mesh name="base" scale="2 2 2" file="../meshes/base_link_scale2.STL" />
    </asset>
    <worldbody>
      <body>
        <body name="object" >
          <geom pos="0 0 0" quat="0.7 -0.7 0 0" mesh="base" type="mesh" solimp="0.998 0.998 0.001" solref="0.001 1" density="1000" friction="1000 1000 1000" contype="0" conaffinity="0" group="1" condim="3"/>
          <geom pos="0 0 0" quat="0.7 -0.7 0 0" mesh="base" type="mesh" solimp="0.998 0.998 0.001" solref="0.001 1" density="1000" friction="1000 1000 1000" group="0" condim="3"/>
        </body>
        <site rgba="0 0 0 0" size="0.005" pos="0 0 -0.082" name="bottom_site"/>
        <site rgba="0 0 0 0" size="0.005" pos="0 0 0.075" name="top_site"/>
        <site rgba="0 0 0 0" size="0.005" pos="0.0254 0.0254 0" name="horizontal_radius_site"/>
      </body>
    </worldbody>

</mujoco>

kevinzakka, 2022-08-09:
Can you attach a minimal XML file with the table and the object so that I can debug locally?

yuvaltassa, 2022-08-10:
Try [this](https://github.com/deepmind/mujoco/issues/422#issuecomment-1205768073).

vaesong, 2022-08-10:
> Can you attach a minimal XML file with the table and the object so that I can debug locally?

Thanks, here are the mesh, and xml files
[import_object.zip](https://github.com/deepmind/mujoco/files/9297017/import_object.zip)
.

kevinzakka, 2022-08-15:
@vaesong Did you try the multiccd option?

vaesong, 2022-08-16:
> @vaesong Did you try the multiccd option?

Yes, I have tried it, but it didn't work.

yuvaltassa, 2022-08-16:
This friction value is absurd. Delete that.
Can you try composing your moving object out of several geoms?

closed 2022-08-20 by yuvaltassa