google-deepmind/mujoco #432: Unable to install python bindings on Ubuntu 18.04.2 LTS

pab47, 2022-08-08:
I have no issues installing python bindings on mac/win, but cannot do so on Ubuntu 18.04.2 LTS (installed via VM). I have python 3.10.6. When I type pip install mujoco I get the message 

"Could  not find a version that satisfies the requirement mujoco (from versions: )
No matching distribution found for mujoco"

From the error, I am not able to understand what I am missing.

saran-t, 2022-08-08:
Can you please post the output of `pip install -vvv mujoco`?

pab47, 2022-08-08:
Thanks for the quick response. I am attaching the output in a text file. Please let me know if the file does not open and/or I need to check anything else. Many thanks.
[mujoco_install_output.txt](https://github.com/deepmind/mujoco/files/9284290/mujoco_install_output.txt)