google-deepmind/mujoco #433: Problem when grasping with an end-effector as a mocap body

fedeceola, 2022-08-09:
Hi, I'm using MuJoCo to simulate a graping task with a humanoid robot. The simulation works really well with the whole model of the robot, but I'm trying to control the end-effector as a mocap body (i.e. removing from my .xml model all the components not included in the hand used for grasping). The problem that I am facing is that, with this configuration, the hand of the robot is not able to lift the object. I have tried to set different masses of the robot's meshes and of the object that I want to grasp, but the problem persists. Do you suggest to modify other parameter in the .xml files of the robot or of the objects?

Here is a video, illustrating my problem:

https://user-images.githubusercontent.com/32268209/183707491-b26cc7eb-9bc3-4e25-b142-d85917130309.mp4

Thanks in advance!

kevinzakka, 2022-08-09:
There's a few things you can try:

* Increase `impratio` by one or two orders of magnitude
* Switch to elliptic friction cones

e.g. `<option cone="elliptic" impratio="100"/>`

There are other things you can do at the level of the geom of your fingers:

* Increase `solimp` (first two values)
* Change `condim`

See [here](https://mujoco.readthedocs.io/en/latest/overview.html#softness-and-slip) for more information.

fedeceola, 2022-08-10:
Hi @kevinzakka, thanks for your response. I tried to modify the parameters as you suggested, but the problem persists. However, the parameters that I'm using are the ones that work with the whole model of the robot. Do you think that there are reasons why I should change these parameters when using only the hand model?

One problem that I can see is that, as mentioned also in #101, mocap bodies "teleport" since they do not have velocity. Do you think that this may cause some problems with contacts computation when lifting the object?

yuvaltassa, 2022-08-10:
Can you just attach a loadable model and we'll take a look?

fedeceola, 2022-08-10:
Hi @yuvaltassa, sure! Here you can find the models of both the hand and the sugarbox.

[hand_sugarbox_model.zip](https://github.com/deepmind/mujoco/files/9299570/hand_sugarbox_model.zip)

Thanks!

yuvaltassa, 2022-08-10:
This is _almost_ helpful :)

Your question is about the hand grasping the box, so please send me a model with the hand and the box, the one you show in the video. [This function](https://github.com/deepmind/dm_control/blob/main/dm_control/mjcf/export_with_assets.py) might prove helpful?

fedeceola, 2022-08-10:
Here's the complete model! Thanks :)

[mjcf_with_assets.zip](https://github.com/deepmind/mujoco/files/9299786/mjcf_with_assets.zip)

yuvaltassa, 2022-08-20:
Thanks for attaching a usable model. If do so in the future you will get a much faster response.

Your issue is that you are putting the mocap body directly on the hand, rather than attaching the hand to the mocap body with a **weld**, as explained and recommended [here](https://mujoco.readthedocs.io/en/latest/modeling.html#mocap-bodies).

I added these lines to your model and everything works fine
```xml
    <body name="arm" mocap="true" pos="0 0 1.5">
      <geom type="box" size=".02 .03 .03" contype="0" conaffinity="0"/>
    </body>
    ...
    <body name="icub_r_hand_sphere" pos="0 0 1.5" quat="0.70710678100000002 0.70710678100000002 0 0">
      <freejoint/>
    ...
  <equality>
    <weld body1="arm" body2="icub_r_hand_sphere"/>
  </equality>    
```

Closing this, let us know if you have further questions.

[image]

closed 2022-08-20 by yuvaltassa

fedeceola, 2022-08-22:
Thanks, now it works. As explained [here](https://mujoco.readthedocs.io/en/latest/modeling.html#mocap-bodies), `icub_r_hand_sphere` was my second body and `icub_r_hand` my actual mocap body, but I added `icub_r_hand_sphere` to the hand and not to the worldbody. Thanks again.