google-deepmind/mujoco #436: Leaking Contacts Through Convex Hulls

tudorjnu, 2022-08-12:
Hi,

I am trying to create an endovascular simulator for RL using MuJoCo. The aorta is decomposed into a series of convex hulls using V-HACD whereas a guidewire is represented as a rope. The idea is to push the guidewire through the aorta using translation and rotation. 

The issue comes when the guidewire (rope) comes in contact with the convex hulls. If the guidewire collides between the hulls, it will penetrate the aorta (see images attached). Is there a way to prevent this from happening?

Thank you!

[image]

[image]

kevinzakka, 2022-08-12:
Can you generate a tighter approximation with V-HACD?

yuvaltassa, 2022-08-12:
The simplest thing would be to overlap the geoms. Geoms belonging to the same body cannot collide with each other and can (and in your case should) overlap. A quick and dirty way to do this would be to add a `margin` to the existing one i.e. "inflate" them. This will have the disadvantage that it will narrow your "aorta" and that the visual geometry won't match the collision geometry, but it should prevent the penetration.

As a second stage you could recreate your decomposition or better yet, construct it procedurally with primitives rather than meshes.

tudorjnu, 2022-08-13:
I have tried to play a bit with the parameters of V-HACD but it does not work as planned. For tiny objects with which I am working, the newer version seems to not work that well as it produces convex hulls with lower than the required volume for qhull. So I refactored obj2mjcf for V-HACD version 3.  

The margin parameter seems to work quite well as it seems the guidewire does not penetrate the aorta anymore. I will try to make the geoms overlap so I can get a tighter approximation. Is there any parameter that can adjust this within V-HACD? 

Thank you very much for the prompt answers!

kevinzakka, 2022-08-13:
@tudorjnu, I have indeed noticed that v4 produces almost flat meshes at times which makes qhull complain. Have you tried giving this [video](https://www.youtube.com/watch?v=Sa1NE-St8iA) a watch?

tudorjnu, 2022-08-13:
@kevinzakka, yes, I watched the video but it doesn't solve the problem. And when I increase the number of the convex hulls enough to have the arteries accessible, most of the time I get a qhull error. 

This is my simplistic version of obj2mjcf using trimesh and V-HACD-3 (in case someone has similar issues): https://gist.github.com/tudorjnu/198ffec88b50810e77b0697e7919fe25 

However, it uses STL (which might be helpful for people that use the earlier versions of gym and mujoco).

closed 2022-08-20 by yuvaltassa

tychen32, 2022-10-20:
Hi, I found this issue related to a problem I'm facing. I converted a cup model using V-HACD-4 and got an obj file containing a collection of 64 convex hulls. However, when I tried to import the converted file into the mujoco scene, it only showed one piece of the convex hulls. What is the right way to import the collection of convex hulls into mujoco?

yuvaltassa, 2022-10-20:
Did you try @kevinzakka's obj2mjcf ?

tychen32, 2022-10-20:
Thanks for the prompt answer! I tried obj2mjcf and was able to import the whole cup model. But it seems mujoco still creates a convex hull for the model and prevents anything from going into the cup. What should I do if I want the cup to behave like a real cup then?

tychen32, 2022-10-21:
I think I've figured it out using obj2mjcf with vhacd. Thank you!

kevinzakka, 2022-10-21:
@tychen32 Please feel free to [raise an issue](https://github.com/kevinzakka/obj2mjcf/issues) if you run into any future problems with `obj2mjcf` :)