google-deepmind/mujoco #440: what is the new substitution of MjViewer from the old mujoco_py?

Bunnyhop-peach, 2022-08-16:
Hi,

I'm a student trying to use MuJoCo for robot arm controller tests.

I'm looking for some help with migrating the mujoco_py code to the deepmind mujoco.

Here is an example code that explains my question:(python)
self._sim = mjp.MjSim(self._model)
self._viewer = mjp.MjViewer(self._sim) if render else None

So, is there any way to have the same interactive window in deepmind mujoco?

Thanks~

yuvaltassa, 2022-08-16:
We hope the long term replacement will be be based on @aftersomemath's WIP at #419 (or similar), will keep you posted as soon as it's ready. 

Cheers

yuvaltassa, 2022-08-20:
I should have mentioned, `dm_control` does have a [Python-based viewer](https://github.com/deepmind/dm_control/tree/main/dm_control/viewer). Did you try using that?

Closing.

closed 2022-08-20 by yuvaltassa

Bunnyhop-peach, 2022-08-20:
> I should have mentioned, `dm_control` does have a [Python-based viewer](https://github.com/deepmind/dm_control/tree/main/dm_control/viewer). Did you try using that?
> 
> Closing.

Not yet, I'll have a look at it, thanks ;-)

yuvaltassa, 2022-11-12:
See #576