google-deepmind/mujoco #442: Pythong binding for mjd_transitionFD

sherilan, 2022-08-17:
Hi,

I'm a student using mujoco for reinforcement learning.

I was happy to see that version [2.2.1](https://github.com/deepmind/mujoco/releases/tag/2.2.1)  added support for differentiation with `mjd_transitionFD`, but so far I've been unable to access it through the python bindings. I figured it should be accessible at the top level of the package, along with most of the other functions declared in the [mujoco.h](https://github.com/deepmind/mujoco/blob/2.2.1/include/mujoco/mujoco.h) header file, but I get an attribute error when I do:

```python
import mujoco
mujoco.mjd_transitionFD
```

Am I misunderstanding how this feature should be accessed, is it not supposed to be available through the python api, or is there a bug somewhere?

Extra details:
- Mujoco was installed through pip (`pip install mujoco==2.2.1`) in a clean virtual environment
- I'm on MacOS Catalina (64 bit intel)
- I get the same result when testing in a docker container on an Ubuntu 18.04 machine

saran-t, 2022-08-17:
This is an oversight on our part, looks like we simply forgot to add these new functions to the Python bindings.

We'll get this fixed for our next release. Sorry about that!

sherilan, 2022-08-17:
Thanks for the quick reply, @saran-t !

kevinzakka, 2022-08-20:
Fixed via 7b7cf0c27affa53f760a4b30765292ad0e57aeff.

closed 2022-08-20 by yuvaltassa

zalo, 2022-08-23:
A patch release for pip incorporating this would be greatly appreciated!

yuvaltassa, 2022-09-24:
Hey @sherilan, @zalo, check out the colab we just added (top of the [changelog](https://mujoco.readthedocs.io/en/latest/changelog.html) as of September 2022). Uses the derivatives for an LQR controller demo.