google-deepmind/mujoco #443: How to place a site in a composite body

traplotz, 2022-08-18:
Hi,
I'm a student and I'm trying to use MuJoCo for a rope simulation.

What I want to do is to pull a rope at its both ends after moving the rope ends with mocap.
The problem is that I cannot seem to place a site at the end link (while there's no problem with the first link),
because a site should be defined while you are defining the body that it is placed in.

I cannot use equality because I'm using the equality for connecting the rope ends with mocap bodies,
 and I'm going to disable the equality eventually after i'm done with the mocaps.

Here is my model:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <light pos="20 20 20" dir="-1 -1 -1" diffuse=".9 .9 .9"/>
    <site name="so1" pos="20 0 0" size="0.02"/>
    <site name="so2" pos="-5 0 0" size="0.02"/>
    
    <body name="CB0" pos="0 0 0">
      <freejoint/>
      <composite type="rope" count="150 1 1" spacing=".1" offset="0 0 0" prefix="C">
        <joint kind="main" damping="100"/>
        <geom type="capsule" size=".05 .03" mass="0.5" rgba=".8 .8 .8 1"/>
      </composite>
      <site name="sw1" pos="15 0 0" size=".02"/>
      <site name="sw2" pos="0 0 0" size=".02"/>
    </body>

    <body name="mc1" pos="15.1 0 0" mocap="true">
      <geom type="sphere" size=".05" rgba=".3 .3 .9 1"/>
    </body>
    <body name="mc2" pos="-.1 0 0" mocap="true">
      <geom type="sphere" size=".05" rgba=".3 .3 .9 1"/>
    </body>
  </worldbody>
  
  <equality>
    <connect body1="mc1" body2="CB149" anchor="-0.05 0 0" solref="-10000 -1000"/>
  </equality>
  <equality>
    <connect body1="mc2" body2="CB0" anchor="0.05 0 0" solref="-10000 -1000"/>
  </equality>
  
  <tendon>
    <spatial name="OW1" width=".01">
      <site site="so1"/>
      <site site="sw1"/>
    </spatial>
    <spatial name="OW2" width=".01">
      <site site="so2"/>
      <site site="sw2"/>
    </spatial>
  </tendon>

  <actuator>
    <motor name="tension1" gear="1000" tendon="OW1"/>
    <motor name="tension2" gear="1000" tendon="OW2"/>
  </actuator>

</mujoco>
```

</details>

You'll see the site "sw1" that i put at the rope end is somewhere down below, 
because the site is defined within the first link("CB0") of the rope which is tilt.

How can I place the site within the last link ("CB149") then?

Thank you very much.

traplotz, 2022-08-18:
I know I can use the "Save xml" and then edit the saved XML code, as mentioned in [https://github.com/deepmind/mujoco/issues/43](https://github.com/deepmind/mujoco/discussions/52) and [https://github.com/deepmind/mujoco/discussions/52](https://github.com/deepmind/mujoco/issues/43) ,
but when my rope has more than 100 links, I cannot load the edited XML due to the "Element nesting is too deep" error.

yuvaltassa, 2022-08-18:
Sadly there is nothing we can do about that until the nesting issue is fixed. 

Perhaps you can ping [our pull request](https://github.com/leethomason/tinyxml2/pull/910) for tinyxml2 and ask them to accept it?

closed 2022-08-18 by yuvaltassa

traplotz, 2022-08-18:
Hi, Yuval.
I think I found a little trick around this problem. I can change the structure of the rope from 

```
<body (first link)>
  <body>
    <body>
      <body>
        <body (last link)>
        </body>
      </body>
    </body>
  </body>
</body>
```

to

```
<body (center link)>
  <body>
    <body (last link)>
    </body>
  </body>
  <body>
    <body (first link)>
    </body>
  </body>
</body>
```
.

This would pretty much halve the element nesting depth.