google-deepmind/mujoco #445: Modelling a 360 degree view lidar sensor to detect obstacles in Mujoco environments

nmohank101, 2022-08-18:
I want to test and compare various state-of-the-art robot navigation algorithms in Mujoco environments. To do so I want to use a 360 degree view [laser range finder ](https://www.slamtec.com/en/Lidar/A1) as the master sensor for the robot to navigate in obstacle free space. Can anyone please let me know, if such a sensor is already modeled and available off the shelf for Mujoco. It would be of great help. 

I agree that [Gazebo simulator](https://classic.gazebosim.org/tutorials?tut=guided_i1) already provides such a feature, but the environment and robot model that I'm working on is built in Mujoco simulator. So, could anyone please help me out. 

I also came across this [safetyGym ](https://github.com/openai/safety-gym) where they've a mention about providing such sensor off the shelf. Not sure whether that would work. If anyone has already worked on this, it would of great help if you can share your experience.

liweiqing97, 2023-03-20:
I also want to make such a design. Have you solved it

liweiqing97, 2023-03-23:
Thank you. That's very helpful to me. I like it, but I do not kown how to fix it