google-deepmind/mujoco #447: Is there a way to make distance equality as strong as possible

Ethanhah, 2022-08-18:
Hi, I am trying to use distance equality to create magnets that I hope to connect two objects as tightly as possible. I guess the only thing left that can adjust the magnetic force is solref and solimp. Is it possible to do that and is there any help please? Thank you.

PS: I also find it is removed from the latest version. Is there any main reason?

yuvaltassa, 2022-08-18:
We removed it because the implementation was janky and we couldn't think of a reasonable use case. What's yours?

Specifically, why are you using a distance rather than a weld?

Ethanhah, 2022-08-20:
So far it looks like kind of stable. Using the feature that it brings two objects together, seems reasonable to be used as magnets (a small attachment to a bigger object). I know some use weld to achieve this, for us we think it is easy to implement and still serves the purpose (cause we kind of restrain the movement on the floor). 

As for the question, I later figure out increasing dmin and dmax seems to do the job.

yuvaltassa, 2022-08-20:
I'm still not convinced you need the distance constraint, but happy to hear you've resolved your issue.

closed 2022-08-20 by yuvaltassa

yuvaltassa, 2022-08-20:
To clarify further, since I failed to understand why your use-case demands a distance constraint (rather than a weld or an [adhesion actuator](https://mujoco.readthedocs.io/en/latest/XMLreference.html#adhesion)), we are currently not reconsidering our decision to remove it.

If you wish us to reconsider, please continue to engage with us and explain why you think you need it.

Thanks