google-deepmind/mujoco #448: Having trouble building from source in Windows 10 ``qhull` - target `qhull` not defined.`

MotorCityCobra, 2022-08-19:
I make a build file in the root directory and run `cmake ..` in the build file. I get the error below.
I am using Visual Studio 2022 and I just updated to make sure that wasn't the problem. I am up to date as of this post. I still have the problem.

```
cmake ..
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.18363.
-- mujoco::FindOrFetch: checking for targets in package `qhull`
-- mujoco::FindOrFetch: checking for targets in package `qhull` - target `qhull` not defined.
-- mujoco::FindOrFetch: Using FetchContent to retrieve `qhull`
MSBuild version 17.3.0+f67e3d35e for .NET Framework
  Checking Build System
  Performing update step for 'qhull-populate'
  Building Custom Rule C:/Users/ooo/tor/mujoco/build/_deps/qhull-subbuild/CMakeLists.txt
  Performing patch step for 'qhull-populate'
CUSTOMBUILD : error : unknown switch `q' [C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\qhull-populate.vcxproj]
  usage: git apply [<options>] [<patch>...]

      --exclude <path>      don't apply changes matching the given path
      --include <path>      apply changes matching the given path
      -p <num>              remove <num> leading slashes from traditional diff paths
      --no-add              ignore additions made by the patch
      --stat                instead of applying the patch, output diffstat for the input
      --numstat             show number of added and deleted lines in decimal notation
      --summary             instead of applying the patch, output a summary for the input
      --check               instead of applying the patch, see if the patch is applicable
      --index               make sure the patch is applicable to the current index
      -N, --intent-to-add   mark new files with `git add --intent-to-add`
      --cached              apply a patch without touching the working tree
      --unsafe-paths        accept a patch that touches outside the working area
      --apply               also apply the patch (use with --stat/--summary/--check)
      -3, --3way            attempt three-way merge if a patch does not apply
      --build-fake-ancestor <file>
                            build a temporary index based on embedded index information
      -z                    paths are separated with NUL character
      -C <n>                ensure at least <n> lines of context match
      --whitespace <action>
                            detect new or modified lines that have whitespace errors
      --ignore-space-change
                            ignore changes in whitespace when finding context
      --ignore-whitespace   ignore changes in whitespace when finding context
      -R, --reverse         apply the patch in reverse
      --unidiff-zero        don't expect at least one line of context
      --reject              leave the rejected hunks in corresponding *.rej files
      --allow-overlap       allow overlapping hunks
      -v, --verbose         be verbose
      --inaccurate-eof      tolerate incorrectly detected missing new-line at the end of file
      --recount             do not trust the line counts in the hunk headers
      --directory <root>    prepend <root> to all filenames

  No configure step for 'qhull-populate'
  No build step for 'qhull-populate'
  No install step for 'qhull-populate'
  No test step for 'qhull-populate'
  Completed 'qhull-populate'
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): error MSB8066: Custom build for 'C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\CMakeFiles\a0d5f46346c3f91cace92e77950e8a23\qhull-populate-update.rule;C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\CMakeFiles\a0d5f46346c3f91cace92e77950e8a23\qhull-populate-patch.rule;C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\CMakeFiles\a0d5f46346c3f91cace92e77950e8a23\qhull-populate-configure.rule;C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\CMakeFiles\a0d5f46346c3f91cace92e77950e8a23\qhull-populate-build.rule;C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\CMakeFiles\a0d5f46346c3f91cace92e77950e8a23\qhull-populate-install.rule;C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\CMakeFiles\a0d5f46346c3f91cace92e77950e8a23\qhull-populate-test.rule;C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\CMakeFiles\251d50f6fc16ac65e693d7e2b2684be4\qhull-populate-complete.rule;C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\CMakeFiles\c70f69066b9d4a0cb448175cab57e292\qhull-populate.rule;C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\CMakeLists.txt' exited with code -1. [C:\Users\ooo\tor\mujoco\build\_deps\qhull-subbuild\qhull-populate.vcxproj]

CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.23/Modules/FetchContent.cmake:1088 (message):
  Build step for qhull failed: 1
Call Stack (most recent call first):
  C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.23/Modules/FetchContent.cmake:1217:EVAL:2 (__FetchContent_directPopulate)
  C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.23/Modules/FetchContent.cmake:1217 (cmake_language)
  cmake/FindOrFetch.cmake:125 (FetchContent_Populate)
  cmake/MujocoDependencies.cmake:107 (findorfetch)
  CMakeLists.txt:53 (include)

-- Configuring incomplete, errors occurred!
See also "C:/Users/ooo/tor/mujoco/build/CMakeFiles/CMakeOutput.log".

```

MotorCityCobra, 2022-09-01:
> @yzd0014 @MotorCityCobra could you please verify the version of Git on your machines for me?

I don't know. I don't have access to the machine right now. I remember I fixed this issue by updating Visual Studio and it worked fine.

saran-t, 2022-09-01:
Reopening since @yzd0014 is still encountering this issue

yzd0014, 2022-09-02:
I try to follow the solution of @MotorCityCobra, but I still got the same error after updating visual studio 2022.

yzd0014, 2022-09-02:
Hey, it turns out that it's indeed a git version problem. After update to the latest git, "git apply -q" becomes valid and the error is gone.

yzd0014, 2022-09-02:
> @yzd0014 @MotorCityCobra could you please verify the version of Git on your machines for me?

Hey, it turns out that it's indeed a git version problem. After update to the latest git, "git apply -q" becomes valid and the error is gone.

saran-t, 2022-09-02:
Thanks for confirming! I'll close the bug, but if you can leave a comment about what Git version had this problem it might be helpful to other users.

closed 2022-09-02 by saran-t

yzd0014, 2022-09-02:
It's git-2.29.2. that has this problem.