google-deepmind/mujoco #449: Issue with large friction: fall into the plane

98Ethan, 2022-08-20:
Hi,

I need to large friction floor to prevent objects from being pushed away. But I find the object would fall into the plane using a large friction (e.g. 8). 

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="base">
    <visual>
        <headlight ambient="0.3 0.3 0.3" diffuse="0.6 0.6 0.6" specular="0 0 0"/>
    </visual>

    <worldbody>
        <geom name="floor" friction="8" type="plane" size=".7 .7 .1" rgba="0.7 0.7 0.7 1"/>
        <body >
            <geom type="box" size="0.1 0.1 0.1" pos="0 0 0.1"/>
            <freejoint/>
        </body>
    </worldbody>
</mujoco>
```

</details>

Here is a screenshon, illustrating my question:
[image]

Thanks!

yuvaltassa, 2022-08-20:
`friction="8"`  is incredibly unrealistic nothing has that friction value. If you want to model the real world, don't do that.

Specifically, this friction value completely breaks the assumptions of the default pyramidal approximation. If you switch to `<option cone="elliptic"/>` you will get no penetration. However the model will still behave in funny ways.

I highly recommend using a realistic friction value (<= 1) and to use the `noslip` or `impratio` options to prevent any minor slippage.

closed 2022-08-20 by yuvaltassa

yuvaltassa, 2022-08-20:
Also, thanks for including a fully loadable model 🙂