google-deepmind/mujoco #450: Non-holonimic contstraints in mujoco. It is possible?

nor-code, 2022-08-21:
Hi there, @yuvaltassa @saran-t @quagla @erez-tom

I'm a phd-student and I'm trying to use MuJoCo for simulation sphere robot.

I'm looking for some help with non-holonimic constraints.

Can i described non-holonimic constraints in mujoco?
For example, there are the spherical shell inside which move a wheel. I want that wheel and shell in contact point have the same velocity.
In other word, wheel must move inside shell without slipping.

And as saying in[ documentation ](https://mujoco.readthedocs.io/en/latest/computation.html?highlight=non-holonomic%20constraints#equality):
"The solver can actually work with non-holonomic constraints as well, but we do not yet have such constraint types defined. "
It turns out that there are no tags to describe such constraints? Is there any way to set these constraints?
I would be very grateful if you could give me an example!

pic and equation(v - velocity) below
[image]
[image]

yuvaltassa, 2022-08-22:
Re general non-holonomic constraints, right now we don't have any.

But I think this specific use case can be implemented as a holonomic constraint?

You can put an equality constraint between the angle $\beta$ and the angle $\phi$ (with the appropriate coefficient) and you'll get the same effect.

nor-code, 2022-08-22:
Please, сan you show how do this? (not found in doc.) 

For plane case non-holonomic equation (pic above) look like this (r and R are constant):
[image]

if considered 3D model, then non-holonomic equations will be:

[image]
Thus the velocities of the generalized coordinates are multiplied by the functions of the generalized coordinates. And in that case, I can set the constraints, as you said?

nor-code, 2022-08-22:
> You can put an equality constraint between the angle β and the angle ϕ (with the appropriate coefficient) and you'll get the same effect.

Please, сan you show how do this? (not found in doc.) 

For plane case non-holonomic equation (pic above) look like this (r and R are constant):
[image]

if considered 3D model, then non-holonomic equations will be:

[image]
Thus the velocities of the generalized coordinates are multiplied by the functions of the generalized coordinates. And in that case, I can set the constraints, as you said?

yuvaltassa, 2022-08-22:
I'm saying the instead of the non-holonomic constraint $\dot{\alpha} + \dot{\beta} - \frac{r}{R}\dot{\phi}=0$ you can implement the holonomic constraint ${\alpha} + {\beta} - \frac{r}{R}{\phi}=0$ using a couple of [joint equality constraints](https://mujoco.readthedocs.io/en/latest/XMLreference.html#equality-joint). This is a stronger statement, but the non-holonomic constraint would of course follow.

Would this accomplish what you want? Let me know if you try and fail, I can help you.

closed 2022-09-11 by yuvaltassa