google-deepmind/mujoco #455: Cannot find GL/gl.h when building from source

vmoens, 2022-08-25:
Hi,

Following this [issue](https://github.com/deepmind/dm_control/issues/310) I'm trying to build mujoco from source, hoping that it will solve my rendering problems.

I have the following libraries installed on the machine:
- libglfw3
- libglew2.0
- libgl1-mesa-glx
- libosmesa6

Here's my workflow:
```
git clone https://github.com/deepmind/mujoco
cd mujoco
mkdir build

conda install -c conda-forge gcc -y
conda install -c conda-forge ninja -y

cmake .. -DCMAKE_BUILD_TYPE:STRING=Release -DCMAKE_INSTALL_PREFIX:STRING=$HOME/.mujoco -DMUJOCO_BUILD_EXAMPLES:BOOL=OFF -DGLFW_USE_OSMESA=1 -DCMAKE_C_COMPILER:STRING=gcc -DCMAKE_CXX_COMPILER:STRING=g++ -G Ninja
cmake --build . --config=Release
```

When calling `cmake --build` I get the following failure on multiple occasions:
```
FAILED: simulate/CMakeFiles/simulate.dir/uitools.c.o
/data/home/vmoens/cluster/conda/envs/mbrl_mujocosource/bin/gcc -DEIGEN_MPL2_ONLY -DMJSIMULATE_STATIC -DmjUSEPLATFORMSIMD -I/data/home/vmoens/cluster/work/mujoco/simulate -I/data/home/vmoens/cluster/work/mujoco/build/_deps/glfw-src/include -I/data/home/vmoens/cluster/work/mujoco/build/_deps/lodepng-src -I/data/home/vmoens/cluster/work/mujoco/include -O3 -DNDEBUG -fPIE -fvisibility=hidden -fdata-sections -ffunction-sections -mavx -Wno-int-in-bool-context -Wno-maybe-uninitialized -Wno-sign-compare -Wno-str
ingop-overflow -Wno-stringop-truncation -pthread -std=c11 -MD -MT simulate/CMakeFiles/simulate.dir/uitools.c.o -MF simulate/CMakeFiles/simulate.dir/uitools.c.o.d -o simulate/CMakeFiles/simulate.dir/uitools.c.o -c /data/home/vmoens/cluster/work/mujoco/sim
ulate/uitools.c
In file included from /data/home/vmoens/cluster/work/mujoco/simulate/uitools.h:18,                                                                                                                                                                                             from /data/home/vmoens/cluster/work/mujoco/simulate/uitools.c:15:
/data/home/vmoens/cluster/work/mujoco/build/_deps/glfw-src/include/GLFW/glfw3.h:241:12: fatal error: GL/gl.h: No such file or directory                                                                                                                         241 |   #include <GL/gl.h>
      |            ^~~~~~~~~                                                                                                                                                                                                                                  compilation terminated.
```

Any help would be appreciated

nimrod-gileadi, 2022-08-26:
On Ubuntu, try installing:
 libosmesa6-dev libxrandr-dev libxinerama-dev libxcursor-dev libegl1-mesa-dev

traversaro, 2022-08-26:
Note that if you are using compilers from conda-forge, you may have unexpected behaviour in using them with the libraries or tools provided by apt, because you could be mixing different ABI-incompatible versions of the same library or the conda-forge compiler can have different default locations for looking for includes or libraries. 

If you want to compile mujoco with apt depencies, there is any reason why you can't use the gcc installed by apt? 

If instead you want to use conda to install the dependencies of mujoco, you probably need to install the `[mesa-libgl-devel-cos7-x86_64](https://anaconda.org/conda-forge/mesa-libgl-devel-cos7-x86_64)` package (see https://conda-forge.org/docs/maintainer/knowledge_base.html#libgl and https://github.com/conda-forge/cdt-builds).

vmoens, 2022-08-26:
Thanks @traversaro 

The main reason why I was using gcc from conda is that I assumed that I needed a newer version than the one installed with apt (7) since the .github workflow uses version >10.
I still have the same issue with the native gcc and the libs mentioned above installed with apt.

The reason why i'm using conda is that i have a limited access to sudo on the clusters i'm working on, and I must rely on other engineers in another time zone to install packages for me (which they're a bit reluctant to do as they will be exposed to the whole team).

@nimrod-gileadi 
I have tried to install them with conda, but it did not work. Again, apt may succeed where conda fails but it's harder for me to install with apt.

saran-t, 2022-08-26:
@traversaro On the other hand, as soon as you start interacting with X11 and GLX you're most likely already interacting with stuff outside of conda-forge anyway, no?

saran-t, 2022-08-26:
@vmoens I'd just try and see if anything goes wrong with older GCC. Just because we only run GCC down to version 9 on GH Actions doesn't mean that things won't work with older versions.

vmoens, 2022-08-26:
Thanks @saran-t 
I tried with gcc 7 but still get the GL/gl.h error.

saran-t, 2022-08-26:
The header lookup issue isn't going to go away by changing the compiler version, no. I was responding to your comment `The main reason why I was using gcc from conda is that I assumed that I needed a newer version than the one installed with apt`.

traversaro, 2022-08-26:
> @traversaro On the other hand, as soon as you start interacting with X11 and GLX you're most likely already interacting with stuff outside of conda-forge anyway, no?

Kind of. Some X11 libraries are packaged as regular libraries in conda-forge, while regarding libGL.so the system one is tipically used at runtime, while at build time a CDT package (https://conda-forge.org/docs/maintainer/knowledge_base.html#core-dependency-tree-packages-cdts) is typically used.

traversaro, 2022-08-26:
(I updated previous comment with a more in-depth explanation).

kevinzakka, 2022-12-06:
Hi @vmoens, is this still an issue?

vmoens, 2022-12-07:
Nope this can be closed as far as I'm concerned

closed 2022-12-07 by vmoens