google-deepmind/mujoco #458: How to make env with gym

yingchengyang, 2022-08-28:
Hi,

I'm a phd student and I'm trying to use MuJoCo for research.

I'm looking for some help with mujoco and gym.

Can I directly use the following instructions?

pip install gym
pip install mujoco

import gym
env = gym.make("Ant-v3")

That's all! Thank you.

kevinzakka, 2022-08-29:
The latest `v4` environments in Gym use the official (aka this repository) Python bindings. So all you need to do is `pip install "gym[mujoco]"`, as described [here](https://github.com/openai/gym#mujoco-environments).

yingchengyang, 2022-08-29:
Thanks a lot!

closed 2022-09-07 by saran-t

BaiYunpeng1949, 2022-12-10:
Hi

I am also a rookie trying to use mujoco + gym. How can I customize my mujoco environment which is inherited from gym.Env and could be used as normal gym environments in reinforcement learning? Should I follow the templates stating 'myEnv' class like [here](https://github.com/openai/gym/blob/master/gym/envs/mujoco/mujoco_env.py) (from line 28)? Or could you please recommend me some examples?

Many thanks!