google-deepmind/mujoco #459: Is it possible to get a reading of how far a body or sensor is from the ground?

MotorCityCobra, 2022-08-29:
Is it possible to get a reading of how far a body or sensor is from the ground?

kevinzakka, 2022-08-29:
The simplest thing is to directly read the Cartesian position of the body by indexing into `MjData.xpos`. See [here](https://mujoco.readthedocs.io/en/latest/APIreference.html#mjdata) for more details.

MotorCityCobra, 2022-08-30:
@kevinzakka Would you happen to know how to return a name of a geom or body or actuator in MjData?  
I'm not sure which is which in my XML model. I could guess, but I'd like to put a tag in an XML and get it printed in the C++.

yuvaltassa, 2022-08-31:
Look at `mj_name2id()` and `mj_id2name`

yuvaltassa, 2022-08-31:
1. The vertical direction in MuJoCo is Z, not Y.
2. `frameyaxis` is the Y-axis of the site's frame, i.e. the second column of the `mjData.site_xmat`. Not what you want.
3. What you want is the [`framepos` sensor](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-framepos), and you want the 3rd (Z) element of this vector.

MotorCityCobra, 2022-08-31:
I'm using `framepos sensor` and printing `std::cout << d->sensordata[2]` but the number is higher when the humanoid's head is lower and vice versa.

```
<!-- Copyright 2021 DeepMind Technologies Limited

     Licensed under the Apache License, Version 2.0 (the "License");
     you may not use this file except in compliance with the License.
     You may obtain a copy of the License at

         http://www.apache.org/licenses/LICENSE-2.0

     Unless required by applicable law or agreed to in writing, software
     distributed under the License is distributed on an "AS IS" BASIS,
     WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
     See the License for the specific language governing permissions and
     limitations under the License.
-->
<mujoco model="Humanoid">
    <compiler inertiafromgeom="true" angle="degree"/>
    <default>
        <joint limited="true" damping="1" armature="0"/>
        <geom condim="1" material="matgeom"/>
        <motor ctrlrange="-.4 .4" ctrllimited="true"/>
    </default>
    <option timestep="0.005" iterations="50" tolerance="1e-10" solver="Newton" jacobian="dense" cone="pyramidal"/>
    <size nconmax="50" njmax="200" nstack="10000"/>
    <visual>
        <map force="0.1" zfar="30"/>
        <rgba haze="0.15 0.25 0.35 1"/>
        <quality shadowsize="2048"/>
        <global offwidth="800" offheight="800"/>
    </visual>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
        <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
        <texture name="texgeom" type="cube" builtin="flat" mark="cross" width="127" height="1278"
            rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" markrgb="1 1 1" random="0.01"/>
        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
        <material name="matgeom" texture="texgeom" texuniform="true" rgba="0.8 0.6 .4 1"/>
    </asset>

    <worldbody>
        <geom name="floor" pos="0 0 0" size="0 0 .25" type="plane" material="matplane" condim="3"/>
        <light directional="false" diffuse=".2 .2 .2" specular="0 0 0" pos="0 0 5" dir="0 0 -1" castshadow="false"/>
        <light mode="targetbodycom" target="torso" directional="false" diffuse=".8 .8 .8" specular="0.3 0.3 0.3" pos="0 0 4.0" dir="0 0 -1"/>
        <body name="torso" pos="0 0 1.4">
            <freejoint name="root"/>
            <geom name="torso1" type="capsule" fromto="0 -.07 0 0 .07 0"  size="0.07"/>
            <geom name="head" type="sphere" pos="0 0 .19" size=".09"/>
            <site name="head0"/>

            <geom name="uwaist" type="capsule" fromto="-.01 -.06 -.12 -.01 .06 -.12" size="0.06"/>
            <body name="lwaist" pos="-.01 0 -0.260" quat="1.000 0 -0.002 0" >
                <geom name="lwaist" type="capsule" fromto="0 -.06 0 0 .06 0"  size="0.06" />
                <joint name="abdomen_z" type="hinge" pos="0 0 0.065" axis="0 0 1" range="-45 45" damping="5" stiffness="20" armature="0.02" />
                <joint name="abdomen_y" type="hinge" pos="0 0 0.065" axis="0 1 0" range="-75 30" damping="5" stiffness="10" armature="0.02" />
                <body name="pelvis" pos="0 0 -0.165" quat="1.000 0 -0.002 0" >
                    <joint name="abdomen_x" type="hinge" pos="0 0 0.1" axis="1 0 0" range="-35 35" damping="5" stiffness="10" armature="0.02" />
                    <geom name="butt" type="capsule" fromto="-.02 -.07 0 -.02 .07 0"  size="0.09" />
                    <body name="right_thigh" pos="0 -0.1 -0.04" >
                        <joint name="right_hip_x" type="hinge" pos="0 0 0" axis="1 0 0" range="-25 5"   damping="5" stiffness="10" armature="0.01" />
                        <joint name="right_hip_z" type="hinge" pos="0 0 0" axis="0 0 1" range="-60 35"  damping="5" stiffness="10" armature="0.01" />
                        <joint name="right_hip_y" type="hinge" pos="0 0 0" axis="0 1 0" range="-120 20" damping="5" stiffness="20" armature="0.01" />
                        <geom name="right_thigh1" type="capsule" fromto="0 0 0 0 0.01 -.34"  size="0.06" />
                        <body name="right_shin" pos="0 0.01 -0.403" >
                            <joint name="right_knee" type="hinge" pos="0 0 .02" axis="0 -1 0" range="-160 -2" stiffness="1" armature="0.0060" />
                            <geom name="right_shin1" type="capsule" fromto="0 0 0 0 0 -.3"   size="0.049" />
                            <body name="right_foot" pos="0 0 -.39" >
                                <joint name="right_ankle_y" type="hinge" pos="0 0 0.08" axis="0 1 0"   range="-50 50" stiffness="4" armature="0.0008" />
[... 53 lines cut ...]
        <motor name="right_hip_y"     gear="600" joint="right_hip_y" />
        <motor name="right_knee"      gear="400" joint="right_knee" />
        <motor name="right_ankle_x"   gear="100" joint="right_ankle_x" />
        <motor name="right_ankle_y"   gear="100" joint="right_ankle_y" />
        <motor name="left_hip_x"      gear="200" joint="left_hip_x" />
        <motor name="left_hip_z"      gear="200" joint="left_hip_z" />
        <motor name="left_hip_y"      gear="600" joint="left_hip_y" />
        <motor name="left_knee"       gear="400" joint="left_knee" />
        <motor name="left_ankle_x"    gear="100" joint="left_ankle_x" />
[... cut ...]
        <motor name="right_shoulder1" gear="100" joint="right_shoulder1" />
        <motor name="right_shoulder2" gear="100" joint="right_shoulder2" />
        <motor name="right_elbow"     gear="200" joint="right_elbow" />
        <motor name="left_shoulder1"  gear="100" joint="left_shoulder1" />
        <motor name="left_shoulder2"  gear="100" joint="left_shoulder2" />
        <motor name="left_elbow"      gear="200" joint="left_elbow" />
    </actuator>

    <sensor>
        <framepos name="head_height" objname="head0" objtype="site"/>
    </sensor>

</mujoco>

```

yuvaltassa, 2022-08-31:
Try putting the site in the same place as the head? (add `pos="0 0 .19"`)

closed 2022-08-31 by yuvaltassa