google-deepmind/mujoco #460: mujoco_up  with  "distutils.errors.CompileError: .......failed with exit code 2"  "distutils.errors.DistutilsExecError:"

liyuanhao-star, 2022-08-29:
Hi,

I'm a student and I'm trying to use MuJoCo for do reinforcement learning in robotics

I'm looking for some help with the question as stated in the question

After I installed the "mujoco_py",when I import this package, this problem occurred:
[image]

saran-t, 2022-08-30:
`mujoco_py` is a third-party package that we do not support, and that project is now dormant. Can you use the `mujoco` package instead?

liyuanhao-star, 2022-08-30:
> `mujoco_py` is a third-party package that we do not support, and that project is now dormant. Can you use the `mujoco` package instead?

Thank you for your answer. I will try your suggestion

closed 2022-09-01 by saran-t