google-deepmind/mujoco #462: How to build Python bindings on Windows

aftersomemath, 2022-08-31:
Hi,

This is to continue the discussion in #419 without getting off topic. Also, I think the answer will be useful to others.

As previously discussed, in order to build the Python bindings on Windows we need to use clang as MSVC does not like some of the files.

Thus far I have:

- Built MuJoCo using clang in Visual studio as normal
- "Installed" the result to `C:\Users\levi\mujoco\out\install\x64-Clang-Release`
- Installed CMake, Python3, and CLang/LLVM to the system and add all of them to path
- Downloaded a released source distribution of the Python bindings from PyPi

Then I tried the following commands based on the suggestion in the comments of this [stackoverflow answer](https://stackoverflow.com/a/51381377).

```
set "CC=clang"
set "MUJOCO_PATH=C:\Users\levi\mujoco\out\install\x64-Clang-Release"
pip wheel -v --build-option "--compiler=unix" mujoco-2.2.1.tar.gz
```

It fails quickly and from the output it look like MSVC is still being used.

What is a better way to do this?

saran-t, 2022-09-01:
This is what pass to `cmake` in our internal Windows build.

```
cmake_args=(
    "-G" "Ninja"
    "-DCMAKE_C_COMPILER:STRING=clang"
    "-DCMAKE_CXX_COMPILER:STRING=clang++"
)
```

aftersomemath, 2022-09-01:
Thanks, that helped a lot. The full steps that worked for me are:

Install/add to path: Clang/LLVM, Ninja, CMake, Python

Run the following:

```
set "MUJOCO_PATH=C:\path\to\binaries"
set "MUJOCO_CMAKE_ARGS=-GNinja -DCMAKE_C_COMPILER:STRING=clang -DCMAKE_CXX_COMPILER:STRING=clang++"
pip wheel mujoco-2.2.1.tar.gz
```

You have to use Ninja else MSVC is used instead of clang.

I had to change `setup.py` since the `cp` does not exist in Windows. I'll submit a PR shortly.

commit e0f6441a (2022-09-01): swap out cp call with shutil.copyfile closes #462
closed 2022-09-01 by copybara-service[bot]