google-deepmind/mujoco #463: How many instances of Mujoco can be run at one time?

MotorCityCobra, 2022-09-01:
By 'instance' I mean how many single model simulations can run simultaneously? Not with a window showing the simulations, but in a bare as possible C++ script only. I'm talking about simulations of the same XML file and same C++ each script each instance, though actions would be different. Would the route be to maybe have multiple `mjModel* m` and `mjData* d` for multiple simulations?  
 
I ask because I am doing Markov / Bellman Equation style RL deep learning training on a MuJoCo simulation and I wonder if it is possible or desirable to train on a multiple instances of MuJoCo simulations.  
Same XML and same C++ but each with their own obs, action, reward, and returning a next state.  
Is this done? Is this advisable?

yuvaltassa, 2022-09-01:
Yes, this is very much advisable.

MuJoCo is thread-safe and is meant to be used by multiple threads. See [testspeed.cc](https://github.com/deepmind/mujoco/blob/main/sample/testspeed.cc) for an example of multiple threads, each with its own `mjData`.

closed 2022-09-01 by yuvaltassa

MotorCityCobra, 2022-09-01:
Thank you.
I see `const int maxthread = 512;`

Are these running on the GPU automatically?  
Does MuJoCo automatically use the GPU if it's available?
Does MuJoCo automatically use _multiple GPUs_ if more than one is available?

MotorCityCobra, 2022-09-03:
@yuvaltassa 
When I run [testspeed.cc](https://github.com/deepmind/mujoco/blob/main/sample/testspeed.cc) my machine is only using one thread.  
I'm printing `nthread` and it's just `1`.  

I see this line toward the middle of the `main` function but nothing like this prints when I compile.  
```
  // print start
  if (nthread>1) {
    std::printf("\nRunning %d steps per thread at dt = %g ...\n\n", nstep, m->opt.timestep);
  } else {
    std::printf("\nRunning %d steps at dt = %g ...\n\n", nstep, m->opt.timestep);
  }
```

I run build and run the script with this line `$ cmake --build . && ./mujoco_gym ../humanoid.xml` and I in the terminal I get...  

```
[ 50%] Building CXX object CMakeFiles/mujoco_gym.dir/speedtest.cpp.o
[100%] Linking CXX executable mujoco_gym
[100%] Built target mujoco_gym
Running 10000 steps at dt = 0.005 ...

 Simulation time      : 0.42 s
 Steps per second     : 24075
 Realtime factor      : 120.38 x
 Time per step        : 0.0415 ms

 Contacts per step    : 10.97
 Constraints per step : 48.64
 Degrees of freedom   : 27
```

I have a 3090 GPU. Does MuJoCo use the GPU automatically? I see in the `testspeed.cc` script it tries to use 512 threads. I'm not sure how to get MuJoCo to use more than one thread and how many to use.  
Thank you.

MotorCityCobra, 2022-09-03:
I commented out the nthread being defined within the main function and put this at the top of the main function.  
```
  int nthread = 42;
  omp_set_dynamic(0);                     // disable dynamic scheduling
  omp_set_num_threads(nthread); 
```
It seems to be running 42 instances of MuJoCo simulations at the same time.  
I still have a question about what is running on the GPU. Is it on the GPU and if it's not can I put it on the GPU?

yuvaltassa, 2022-09-03:
`testspeed` is a command line utility. You're supposed to specify the number of threads in the command line. Type `testspeed` with no arguments to read the help documentation.

saran-t, 2022-09-03:
MuJoCo does not run on GPUs.