google-deepmind/mujoco #464: Could we reduce local gravity in mujoco environment?

wangjunyi9999, 2022-09-01:
Hello, 
I am looking for some help with the mujoco env.  Now in my project,  looking at the below image, I want to make the right pink target would reset its positon if the tip hits it.
[image]
In order to achieve it, I use below "target code" 
`  <body name="target" pos="0 0.7 0" euler="0 0 0">
            <joint armature="0" axis="0 0 1" damping="0" limited="true" name="target_z" pos="0 0 0" range="-1 1" ref="0" stiffness="0"  type="slide" />
            <joint armature="0" axis="0 1 0" damping="0" limited="true" name="target_y" pos="0 0 0"  range="-0.2 0.2" ref="0" stiffness="0" type="slide" />
            <joint armature="0" axis="0 0 0" damping="0" limited="true" name="target_x" pos="0 0 0"  range="-1 1" ref="0" stiffness="0"  type="slide"/>
            <geom conaffinity="0" contype="0" name="target" pos="0 0 0" rgba="0.9 0.4 0.6 1" size=".01" type="sphere"/>
        </body>`

But in my mujoco environment, the direction of gravity points down. Therefore, if my target moves, it would always drop to the lowest range of z axis, for example '-1' in my "target code" . I try to set the mass of body is zero, but an error told me it is not allowed : (

Also, if I do not set the joint type is slide, the target would be fixed. I tried use only 'site' instead of 'target body' while it seems worse that the site cannot move.

Really hope someone could help me in this question, you could raise a good solution or idea, either is ok. :) 
Thanks in advance!

yuvaltassa, 2022-09-01:
It is not entirely clear what you are trying to do.

There is a thing called a [mocap body](https://mujoco.readthedocs.io/en/latest/modeling.html#mocap-bodies) (`<body mocap="true">`) which you can move around interactively or programmatically. 

You can have a slider joint with a position actuator that will keep it place.

Maybe you want something else, like I said it is not clear what you are trying to do.

wangjunyi9999, 2022-09-01:
Sorry,  it makes you confused. What I try to do is **resetting the pink target ball position**. I use the target code which I sent last time, but it always **drop to the bottom because of the gravity**. Right now, I only have one idea to limit its position, it is cancelling the joint type "slide". But if so, the target would be fixed.  

By adding what you mentioned before, it seems not what I need, but still thank you! : )

yuvaltassa, 2022-09-01:
So you want the ball to be able to fall, but you only want it to fall when you tell it to fall, and not before, is that correct?

wangjunyi9999, 2022-09-01:
Exactly!

yuvaltassa, 2022-09-01:
Then I think what you want is to apply a force of negative weight (gravity compensation) directly to the ball center of mass.

You can do this by writing into `data.xfrc_applied[i, 2] = m*g` where `i` is the index of the body, 2 is the Z direction.

This looks a bit different in Python or C, I'm not sure which you are using.

wangjunyi9999, 2022-09-01:
I use python, maybe I should define a function in mujoco_env.py?

yuvaltassa, 2022-09-01:
Are you using the native Python bindings or something else?

wangjunyi9999, 2022-09-01:
Oh, that is GYM mujoco environment.

wangjunyi9999, 2022-09-01:
Yes,  thanks for your help. At least now I know where to modify the model.

wangjunyi9999, 2022-09-01:
Great, it works! But there is also a small question, when I use the values of 'm*g' is '0.0042*9.81', the target would float itself a bit. It seems that the force which I set is more than the gravity. However, in .xml file, the gravity is -9.81, and after I print the model, the mass of target is also 0.0042. So I have to twick the mass, and decrease it to 0.004189.

It looks weird. I do not know which one value is not correct. As we know, 'g' is not actual 9.81, it is a bit smaller than the value.  Maybe this is cause, but it is so interesting.

yuvaltassa, 2022-09-01:
This is happening because you a re copying numbers by hand. If you do it procedurally there should be no drift `norm(model.opt.gravity)*model.body_mass[body_id]`.

closed 2022-09-01 by yuvaltassa