google-deepmind/mujoco #465: How do you handle time increments in the simulation without making the C++ wait?

MotorCityCobra, 2022-09-01:
I have my C++, which is running an XML model in MuJoCo waiting every 1/60th of a second to take a step.  
I don't need to look at a simulation in a window, I just want a realistic simulation. How would I step the simulation every 1/60th of a second without having to wait 1/60th of a second but not taking observations and actions in the simulation faster than 1/60th of a second by its timeline?

```
        while (sample_episode_maxstep > 1)
        {
            mjtNum simstart = d->time;
            while (d->time - simstart < 1.0/60.0)
            {
                sample_episode_maxstep--;
```

nimrod-gileadi, 2022-09-07:
I don't understand your question.

1. Do you mean you want to run 1/60th of a second worth of time *in the simulated world* before changing controls and counting a "step"?
2. Or are you talking about simulating as much as possible in 1/60th of a second of real world time?

MotorCityCobra, 2022-09-07:
@nimrod-gileadi Hello.  
I want to do 1. 1/60th of a second in the simulated world.  
Maybe I can ask this for my question... how many units of `d->time` is one second?

MotorCityCobra, 2022-09-08:
@nimrod-gileadi Thank you.  
Let me see if I have this... `m->opt.timestep` is the smallest unit of time in the simulator? 1/200th?  
And when calling a single new step, (in your bottom block of code), we step the simulator those 1/200ths until we are at a step we desire? For instance, if we wanted actions and observations at each second we would step the simulation 200 times between every observation and action.
Is this right? 

A few questions, if you would continue to be so gracious...  
What is the 0.5 in the line `int nstep = (control_step / m->opt.timestep + 0.5);`? Maybe because it's morning, this makes no sense to me.  

1/60th is 0.0166666666666. Is it a problem, or less than ideal that the increment I want is not an integer?

nimrod-gileadi, 2022-09-08:
> Let me see if I have this... m->opt.timestep is the smallest unit of time in the simulator? 1/200th?

Yes, and it's a number you choose through the `<option>` [tag](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option) in your XML.

> What is the 0.5 in the line int nstep = (control_step / m->opt.timestep + 0.5);

This is a C trick for rounding a floating point number to the nearest integer. If I have a `double` called `a`, converting it to int will round it down. Adding `0.5` before rounding down results in rounding to the nearest integer instead.

> 1/60th is 0.0166666666666. Is it a problem, or less than ideal that the increment I want is not an integer?

It doesn't really matter that it's not an integer but it would be convenient to have it as an integer multiple of the simulation timestep (because your code would be simpler, as above).

closed 2023-04-20 by nimrod-gileadi