google-deepmind/mujoco #475: Visualize Multiple robots in the same Window

dhruvthanki, 2022-09-08:
Hi,

I am trying to visualize n robots, all defined using the same XML in the same glfw window using the python API. How can I go about doing this?

Any help would be valuable!
Thank you!

rohanpsingh, 2022-09-09:
Hi @dhruvthanki 
Do you mean you want to _simulate_ multiple robots simultaneously? I do it by merging all my XMLs into one and then loading it up. Then I can do something like [this](https://user-images.githubusercontent.com/16384313/188832982-1a1263e4-ce33-4dc7-804a-589d151670b3.png)!

yuvaltassa, 2022-09-09:
That is indeed how it's done. You can do this manually like @rohanpsingh or with [PyMJCF](https://github.com/deepmind/dm_control/blob/main/dm_control/mjcf/README.md).
The native [`include`](https://mujoco.readthedocs.io/en/latest/XMLreference.html#include) MJCF element is currently rather weak (we are working to improve it), but is sometimes also useful.

closed 2022-09-09 by yuvaltassa

dhruvthanki, 2022-09-10:
Thanks, @yuvaltassa, and @rohanpsingh!
I thought of that but just too many models with the same body, geom, and site names! (@rohanpsingh). I guess PyMJCF is the way to go. But how do I attach multiple robots with the same XML to the same scene?

yuvaltassa, 2022-09-11:
PyMJCF takes care of namespacing for you. Just attach your robots and everything should work.

rohanpsingh, 2022-09-12:
@yuvaltassa For the same purpose, is there a plan to similarly introduce a "namespace" attribute for the `include` MJCF element in the future? Do you think it will be useful (say, to include the same object multiple times in a scene)?

yuvaltassa, 2022-09-14:
Yup, we completely agree :slightly_smiling_face: