google-deepmind/mujoco #478: Ground Reaction Force in the Unity plugin

Balint-H, 2022-09-11:
Hello,

I'm using the Unity plugin of MuJoCo for human gait synthesis. I'm interested in reading out the ground reaction forces. I only need the ground normal / global up component. I can use a touch sensor on the ground, but I would like to have separate signals for the left and right sides.

- Windows 10
- Unity 2021.2.19
- MuJoCo 2.2.0

I figured that `data->cfrc_ext` can be used for this, by summing up the toe and foot body contact forces, with guidance from the old forum. [link_1](https://roboti.us/forum/index.php?threads/does-d-cfrc_ext-give-torque-force-or-opposite.3368/), [link_2](https://roboti.us/forum/index.php?threads/comparing-two-methods-for-obtaining-the-contact-force-on-the-object.3372/#post-3636).

However every time I try to index `cfrc_ext` I get a null reference exception. I'm using the same method I've used to access other members of 'mjData', never ran into this issue before. I tried accessing it at different points of the timestep, same issue, and calling `mj_rnePostConstraint` didn't help either.

My next plan is to use `mj_contactForce` instead, and convert to a global frame of reference as recommended [here](https://github.com/deepmind/mujoco/issues/132#issuecomment-1229432247).

Was using `cfrc_ext` an appropriate approach, just done incorrectly? Or is it an issue with the plugin? Any other alternatives for this purpose?

Many thanks!

yuvaltassa, 2022-09-11:
Won't touch sensors on the _feet_ accomplish you what you want?

Balint-H, 2022-09-12:
I tried that as well, but I was uncertain about a couple of things, so I thought about alternatives. `cfrc_ext` seemed at first to be a quick way to read what I need, and its alignment with the global frame seemed convenient in case I move to sloped surfaces and still need the vertical component.

However, following your recommendation I'll look further into touch sensor based solutions.

I still need to add a second touch sensor to the toes and sum the signal, right (or other bodies of the foot that might contact the ground)?

When working with sloped surfaces I plan to use `mj_contactForce`, or use the normal of the contact.

Thanks for the quick responses!

closed 2022-09-29 by Balint-H