google-deepmind/mujoco #479: Instant crash with MuJoCo 2.2.2 and Unity plugin

Balint-H, 2022-09-11:
Hello,

I'm using the Unity plugin with the following system:

-Windows 10
-Unity 2021.2.19
-MuJoCo 2.2.0

I wanted to upgrade to 2.2.2, but after importing the new version of the package, and the dll, the editor crashes immediately when launching any, even minimal scene. Surprisingly, this doesn't even trigger the Unity editor's usual crash reporting functionality, the application just quits suddenly and silently without hanging.

I tried restarting the editor and my computer. I tried with a fresh project with the same issues. I reverted back to 2.2.0 successfully with the same method I switched to 2.2.2, and scenes work again.

Please let me know if I can provide any further information that may help with this. I'd like to keep up with the updates to the engine!

erez-tom, 2022-09-12:
I reproduced the crash with 2021.3.9.  What happens if you use a binary that was built from code on your own system (as opposed to the one you download from GitHub)?

Balint-H, 2022-09-12:
I built from source, and tested with simulate.exe, the installation looks fine. However, the crash remains when using the new dll in the Unity plugin. I confirmed the unity package files use version 2.2.2 (e.g. in the bindings).

erez-tom, 2022-09-13:
I built a standalone Player executable and ran it from within Visual Studio.  I got an error message "A heap has been corrupted", but I was then able to proceed to the Game, which looked fine...
I tried to use WinDBG, but so far I wasn't able to learn anything from that.
I found that it also happens with V2.2.1 (but not 2.2.0, as you mentioned).  My next guess is that this is due to change 12 [here](https://mujoco.readthedocs.io/en/latest/changelog.html#version-2-2-1-july-18-2022):
Arrays in mjData’s buffer now align to 64-byte boundaries rather than 8-byte.

DinoDany, 2022-09-15:
I'm having the same issue in MAC with M1, working with Unity 2021.3.1f <Rosetta>. It crashes when I try to import a scene.

Balint-H, 2022-09-15:
Do you also get a crash when running a scene built in the editor, as opposed to importing one?

saran-t, 2022-09-15:
A fix is incoming.

DinoDany, 2022-09-15:
@Balint-H No, it only happens when I try to import a MuJoCo scene. It actually didn't work when I tried to revert the version. Maybe it is something else?

closed 2022-09-15 by copybara-service[bot]

saran-t, 2022-09-15:
Could you please try out https://github.com/deepmind/mujoco/commit/68718e76b1b3e04ed3a3db2d6036753d93e26a83 to see if it resolves the crash for you?

DinoDany, 2022-09-15:
Mine still crashes while trying to import a MuJoCo scene :( but I can run a scene built in the editor after importing the package to I'm not sure it is the same problem as @Balint-H

erez-tom, 2022-09-15:
A crash in the import process is probably something different.  If you could trim down your XML to a minimal reproducible example of the problem, I'd be happy to take a look (in a new issue).

Balint-H, 2022-09-16:
> Could you please try out [68718e7](https://github.com/deepmind/mujoco/commit/68718e76b1b3e04ed3a3db2d6036753d93e26a83) to see if it resolves the crash for you?

I can confirm the fix allows the use of the new dll, and scenes run as expected now. Thanks for the quick fix!

commit 68718e76 (2022-09-15): Ensure that memory allocated by `mju_malloc` is freed by `mju_free`.

Previously, `vfs->filedata` memory was allocated via `mju_malloc` in C and released using `Marshal.FreeHGlobal` in C#. In 5cc73c30bfdc34e7b0f612e2ba25a7cdfc4a19ba, `mju_malloc` started returning 64-byte aligned pointers, which should not be freed by a deallocator that isn't aware of this alignment.

Fixes #479.

PiperOrigin-RevId: 474596441
Change-Id: I8a8383c7eaa2c1ec516d81f92a2f90e7aedfe2ee