google-deepmind/mujoco #484: Unity crashing while importing MuJoCo scene

DinoDany, 2022-09-15:
Hi, 

I'm an engineer and I'm trying to use MuJoCo for simulating a robot finger in Unity. I have a .XML file that I had been using on another computer with PopOS and it works ok. Now I'm trying to use my MAC (M1). I can import the MuJco package through the package manager, and I can run the Unity scene just a terrain but when I try to import the MuJoCo scene. It crashes, the whole window closes and it asks me for a bug report.

I attached my model
 
[model.zip](https://github.com/deepmind/mujoco/files/9578092/model.zip)

Thank you so much :)

DinoDany, 2022-09-15:
I'm using MuJoCo 2.2.2

saran-t, 2022-09-15:
Are you using the x86 build of Unity?

DinoDany, 2022-09-15:
Oh, yes. I am.

saran-t, 2022-09-15:
Rosetta2 cannot translate AVX instructions. You should be fine with the Apple Silicon version of Unity.

DinoDany, 2022-09-15:
Thank you so much!

erez-tom, 2022-09-15:
I get a similar result on 2021.3.9 (Silicon) - the Editor freezes when trying to import the meshes.  If I remove the meshes from the file, it loads without a problem.  I see in your XML that you use the same mesh for both collision and visualization.  I would guess that these meshes are too big to be useful for collision...  Also, the meshes look very concave, but remember that MuJoCo creates a convex hull before using the mesh, so none of this detail is even used for the physics itself.  Maybe try to subsample the meshes and create the convex hull, or, if the holes are important, decompose them into convex parts?
I'm re-opening the issue as I suspect it's a problem of the STL importer.  Please update us -- if the problem persists with smaller meshes, we can look into it in more detail.

erez-tom, 2022-09-15:
I managed to import the meshes in Unity by dragging them into the Assets folder in the Project window.  I was then able to add them to the model in the Editor.  This revealed that your meshes are perhaps not centered?  It seems like all meshes share the same global frame origin.  Perhaps it would make more sense to edit each mesh and move it to the origin in its file (BTW, under the hood the MuJoCo compiler does these operations).  This will allow you to author the model in the Unity Editor more easily, and avoid the very specific position specification in the XML.

DinoDany, 2022-09-15:
Hey! Thank you. I was actually able to import the model without any modification using the last version of Unity (Sillicon 2023.1.0a10). It works pretty well. It imported the meshes and I'm able to see a very detailed model. One of the inertia boxes was a little bit out of place but that's all. Nevertheless, I will totally try to avoid very specific position specifications :)

closed 2022-09-16 by erez-tom