google-deepmind/mujoco #485: Mujoco How to apply external force

kpydv, 2022-09-16:
**How to apply external force in mujoco-py at the joint**

Hi,

I want to apply cartesian force at a particular joint in mujoco-py.
can anybody help me out??

saran-t, 2022-09-16:
You mentioned "mujoco-py" in your question. This is the old, third-party Python bindings that are no longer maintained by the authors.

I also saw your post on openai/mujoco-py suggesting that you're trying to use MuJoCo 1.50. If you're a new user I'd strongly encourage you to use the latest version instead (installable via `pip install mujoco`).

In any case, could you expand on what you meant by applying Cartesian force "at a joint"?

kpydv, 2022-09-16:
[image]
Hi, 
currently, I m using the mujoco210 version.
In the above figure, I want to apply a horizontal force in the x direction at the first joint.
hope you understand the problem.

yuvaltassa, 2022-09-16:
You can apply external Cartesian forces to body CoMs using `xfrc_applied`, or at a site using site transmission. Check the documentation for details.

closed 2022-09-16 by yuvaltassa

kpydv, 2022-09-17:
I checked the documentation I did not find any way to apply xfrc_applied at the site.
Please help me out.