google-deepmind/mujoco #486: Unity plugin, Weld offset field leads to instability

Balint-H, 2022-09-16:
Hello,

I'm using Unity 2021.2.19, Windows 10, MuJoCo 2.2.2, and latest version of the plugin. I've been using "puppeteering" from reference animations using mocap bodies and weld constraints. It used to be (back with 2.2.0) that if no offset was provided to the `MjWeld` component, it would use the transformation between the two constrained bodies at startup as the offset. If this was not the desired behaviour, we could supply a dummy, identity transform in the serialized field, so no offset was applied.

 Now if an offset is supplied, even if it is a transform with no `localPosition` or `localRotation` effects, the constraint leads to erratic movements from the constrained body, and very quickly an exception due to instability.
 
 8ca5887c20d47f7534c9c22b3e363b64e4e8e36b added additional data stored in `model->eq_data`, so is this caused by `MjTransformAtEntry` in the C# component writing to the wrong addresses now?

Balint-H, 2022-09-16:
Wanted to add, that if no Weld Offset transform is supplied then no weird behaviour happens perhaps because then edits to `eq_data` are skipped. However this disables the option to have no offset between constrained objects, but start at different states, which can be useful particularly for puppeteering (where the mujoco body definition might be more convenient in a non T-pose for e.g. joint limits, but have the reference animation skeleton in a T-pose).

erez-tom, 2022-09-17:
Your analysis is correct.  I believe you can fix things by replacing  `MjEngineTool.MjTransformAtEntry(model->eq_data, MujocoId)` with `model->eq_data + MujocoId * 11 + 3` (11 being the new value of `mjNEQDATA` following #8ca5887).  Let me know if that works out for you.

kevinzakka, 2022-12-06:
Fixed by #491.

closed 2022-12-06 by kevinzakka