google-deepmind/mujoco #487: Illegal instruction (core dumped) on Ubuntu 20.04, AVX2

xRiis, 2022-09-16:
Hi all, it looks like Mujoco is unusable on my machine. Error is in the title: Illegal instruction (core dumped)

OS: Ubuntu 20.04.5 (running through VirtualBox)
CPU: i9-12900K
Using Mujoco 2.2.1

This occurs as soon as I try to import Mujoco in a Jupyter notebook. I cannot manage to get Mujoco working on the latest Win11 release, either. I have tried installing Mujoco both by pip and from source, in case there was some issue with the package not properly recognizing my CPU architecture.

Our hypothesis is that there's some issue caused by my processor's asm being AVX2 rather than AVX.

saran-t, 2022-09-16:
Check out https://stackoverflow.com/questions/65780506/how-to-enable-avx-avx2-in-virtualbox-6-1-16-with-ubuntu-20-04-64bit

xRiis, 2022-09-16:
No luck, unfortunately. Tried it with the VM I was using as well as a new one (same OS)

saran-t, 2022-09-16:
Just to confirm, if you `cat /proc/cpuinfo` in the VM, does it show AVX and/or AVX2 support?

xRiis, 2022-09-16:
It does not -- will figure out what's going wrong on my end.

saran-t, 2022-09-16:
If you can't get AVX enabled on the VM you can always build MuJoCo from source with AVX disabled.

xRiis, 2022-09-19:
I've followed the StackOverflow solution (as well as the potential solution described in the problem) and despite successfully disabling Hyper-V I have not been able to get my VM access to AVX/AVX2. Any advice?

saran-t, 2022-09-21:
At this point I think you should just build everything (MuJoCo and Python bindings) from source.

kevinzakka, 2022-12-06:
Hi @xRiis, were you able to solve things via Saran’s suggestion?

duhaoyuan1111, 2023-01-14:
> I've followed the StackOverflow solution (as well as the potential solution described in the problem) and despite successfully disabling Hyper-V I have not been able to get my VM access to AVX/AVX2. Any advice?

I tried the something twice, and I met the same issue. I didn't find how to build MuJoCo from source with AVX disabled. May I have a link to the steps? I appreciate that.

nimrod-gileadi, 2023-04-20:
Sorry for the late response. When using CMake, you should be able to specify `-DMUJOCO_ENABLE_AVX=OFF` (see the [config](https://github.com/deepmind/mujoco/blob/08027b4/cmake/MujocoOptions.cmake#L69))

closed 2023-04-20 by nimrod-gileadi

Parker-McDonnell, 2023-12-04:
Did anyone manage to recompile Mujoco and the python bindings from source with AVX disabled? I'm having the same problem trying to run Ubuntu in Virtualbox.

KnDellx956669, 2024-12-02:
> Did anyone manage to recompile Mujoco and the python bindings from source with AVX disabled? I'm having the same problem trying to run Ubuntu in Virtualbox.

I only recompile Mujoco from source but have no idea how to deal with the python bindings with AVX disabled

KnDellx956669, 2024-12-02:
> If you can't get AVX enabled on the VM you can always build MuJoCo from source with AVX disabled.
I just install mujoco simulator based on C++ with AVX disabled but fail to build the python bindings since it will always show "illegal instruction(core dumped)" when i execute "import mujoco"

mujoco==3.2.5

i just follow the "building from resource" part of the documentation to build the python bindings but still fail.....

Sudouble, 2025-01-05:
> Did anyone manage to recompile Mujoco and the python bindings from source with AVX disabled? I'm having the same problem trying to run Ubuntu in Virtualbox.

I tried to disable the AVX, this works for Mujoco in VM

yuvaltassa, 2025-01-06:
Can someone who is failing to compile the bindings without AVX actually say what steps they took and what the failure is?

Sudouble, 2025-01-06:
来件已收到，我会尽快处理！

Mohamed-Abduljawad, 2025-09-08:
I faced the same problem, and I followed the instruction mentioned here, and in the building pages of python and Mujoco app, but still no use. 

So for the mujoco building I did everything mentioned here: https://mujoco.readthedocs.io/en/latest/programming/index.html#inbuild 
Just except for ```cmake $PATH_TO_CLONED_REPO``` I did ```cmake $PATH_TO_CLONED_REPO -DMUJOCO_ENABLE_AVX=OFF ``` as suggested https://github.com/google-deepmind/mujoco/issues/487#issuecomment-1516461811

And for the python bindings I followed the exact instructions in this section: https://mujoco.readthedocs.io/en/latest/python.html#building-from-source

I made sure I uninstalled the previous pypi version, but still whenever I import mujoco I get the error. 
```
>>> import mujoco
Illegal instruction (core dumped)
``` 

Are there more cmake flags to turn off?

Sudouble, 2025-09-08:
来件已收到，我会尽快处理！

yuvaltassa, 2025-09-15:
There is MUJOCO_ENABLE_AVX but also [MUJOCO_ENABLE_AVX_INTRINSICS](https://github.com/google-deepmind/mujoco/blob/4fc69fa64d0a623ccdc4f811a7b9bdb626ae5159/cmake/MujocoOptions.cmake#L70), did you try disabling that too?