google-deepmind/mujoco #490: Here's how to do it with a site transmission. This is explained [here](https://mujoco.readthedocs.io/en/latest/computation.html#transmission), example model [here](https://github.com/deepmind/mujoco/blob/main/test/engine/testdata/refsite.xml).

kpydv, 2022-09-17:
Here's how to do it with a site transmission. This is explained [here](https://mujoco.readthedocs.io/en/latest/computation.html#transmission), example model [here](https://github.com/deepmind/mujoco/blob/main/test/engine/testdata/refsite.xml).

Put a site `force_site` where you want to apply your force. Put a reference site `reference_site` say at the origin of the world. Then if you make an actuator like

```xml
<actuator>
  <motor name="y"  site="force_site" refsite="reference_site" gear="0 1 0 0 0 0"/>
</actuator>
```

the actuator will apply forces in the `Y` axis (of the reference site).

_Originally posted by @yuvaltassa in https://github.com/deepmind/mujoco/issues/489#issuecomment-1250047467_

is it any way to apply externally as disturbance or impact??

yuvaltassa, 2022-09-17:
Please edit the title of your issue before posting. Try to follow [these instructions](https://github.com/deepmind/mujoco#asking-questions) regarding questions on GitHub.

Regarding you question: I don't understand it. What do you mean by "disturbance or impact"?

kpydv, 2022-09-17:
Put a site force_site where you want to apply your force. Put a reference site reference_site say at the origin of the world. Then if you make an actuator like

<actuator>
  <motor name="y"  site="force_site" refsite="reference_site" gear="0 1 0 0 0 0"/>
</actuator>

As you suggested, apply the force at the joint in the XML file.
But I want to apply at force through xfrc_applied to site reference in the main file. (1st question)
can I use that force as an external disturbance for a short during like someone is trying to push and leave? (2nd question)

yuvaltassa, 2022-09-17:
1. What do you mean "in the main file"? The actuator I described above does exactly what you are asking for.
You can use it procedurally by setting `data->ctrl[actuator_id]` instead of some value in `xfrc_applied`. It does the same thing.

2. Yes.

closed 2022-09-17 by yuvaltassa