google-deepmind/mujoco #493: Save and Restore Environment State

Saptam04, 2022-09-20:
I'm trying to restore the environment to a previously seen state. Since mujoco does not support the `get_state()` and `set_state(state)` functions anymore (as explained in [the documentation](https://mujoco.readthedocs.io/en/latest/python.html#migration-notes-for-mujoco-py)), we have to set the relevant fields explicitly. But the observation that I get after restoring the state of the environment is very slightly different from the observation before restoring. Is this the correct way of doing it?

I'm using the following code:
```
import numpy as np
import gym

env = gym.make("Ant-v4")

seed = 100
obs = env.reset(seed = seed)
env.action_space.seed(seed)

nq, nv = env.unwrapped.model.nq, env.unwrapped.model.nv
obs_hidden = env.unwrapped.state_vector()

act = env.action_space.sample()
obs_1, _, _, _ = env.step(act)
print(obs_1)

env.unwrapped.set_state(obs_hidden[: nq], obs_hidden[nq : nq + nv])
obs_2, _, _, _ = env.step(act)
print(obs_2)

print(np.array_equal(obs_1, obs_2))
print(obs_1[24])
print(obs_2[24])
```
Following is a sample output:
```
[  0.78102192   0.99446227   0.06098776   0.07400553   0.04299374
  -0.07730379   0.47748963   0.05383492  -0.38538979   0.1836153
  -0.48892288  -0.12608098   0.47900775   0.36596394   0.3628452
   0.46128181  -0.37403918  -0.8717751   -0.94878363   0.05584823
  13.62777278   4.9388784   -5.24978601   4.30822419 -12.55063306
  -3.66821107  10.81313906]
[  0.78102192   0.99446227   0.06098776   0.07400553   0.04299374
  -0.07730379   0.47748963   0.05383492  -0.38538979   0.1836153
  -0.48892288  -0.12608098   0.47900775   0.36596394   0.3628452
   0.46128181  -0.37403918  -0.8717751   -0.94878363   0.05584823
  13.62777278   4.9388784   -5.24978601   4.30822419 -12.55063306
  -3.66821107  10.81313906]
False
-12.550633062180527
-12.550633062180529
```
Build info:
```
gym                           0.25.2
mujoco                        2.2.2
```

yuvaltassa, 2022-09-20:
You probably need to set the warmstart accelerations. [This](https://mujoco.readthedocs.io/en/latest/computation.html#the-state) is the relevant docs section.

closed 2022-09-20 by yuvaltassa

Saptam04, 2022-09-20:
> You probably need to set the warmstart accelerations. [This](https://mujoco.readthedocs.io/en/latest/computation.html#the-state) is the relevant docs section.

Do you mean to say we need to keep track of `env.unwrapped.data.qacc_warmstart` and set it prior to the second `step`ing? It doesn't seem to change the output.

yuvaltassa, 2022-09-20:
Please can you paste the code where you try this and it fails?

Saptam04, 2022-09-21:
Here is the code:
```
import numpy as np
import gym

env = gym.make("Ant-v4")

seed = 100
obs = env.reset(seed = seed)
env.action_space.seed(seed)

nq, nv = env.unwrapped.model.nq, env.unwrapped.model.nv
obs_hidden = env.unwrapped.state_vector()
acc_hidden = env.unwrapped.data.qacc_warmstart

act = env.action_space.sample()
obs_1, _, _, _ = env.step(act)
print(obs_1)

env.unwrapped.set_state(obs_hidden[: nq], obs_hidden[nq : nq + nv])
env.unwrapped.data.qacc_warmstart = acc_hidden
obs_2, _, _, _ = env.step(act)
print(obs_2)

print(np.array_equal(obs_1, obs_2))
```

yuvaltassa, 2022-09-23:
I tried to figure this out by looking at the gym code but couldn't see anything obviously suspicous.

Since we don't support `mujoco-py` I'm afraid I have no way to truly get to the bottom of this, sorry.

Let me know if you figure it out.