google-deepmind/mujoco #494: Particle Generator

tudorjnu, 2022-09-20:
Hello,

I'm a student using MuJoCo for simulating endovascular interventions (i.e. navigating guidewires and catheters through vasculature).

One of MuJoCo's composite macros is to generate particles. I was wondering if, with the current MuJoCo architecture, it is possible to generate particles at one specific location with a set velocity such that it can resemble fluid (e.g. adding a particle generator somewhere within the library). Indeed there are more considerations but firstly I was wondering if this would, in theory, be feasible.

Thank you very much for your time and for the amount of work that is dedicated to this project!

yuvaltassa, 2022-09-20:
Before getting into high-DoF fluid modeling, did you try the DoF-less fluid model?
There is a `wind` parameter that effectively sets the velocity of the fluid.
If you think this is something you might use, I can make you and example model.
If you need the velocity field to be position-dependent, then `wind` might not be good enough, as it is a global thing.

tudorjnu, 2022-09-20:
Thank you for the response!

Indeed, it has to be position-dependent. Particularly, it has to be contained within the aortic model and flow through the aortic arch which is why I think I am constrained to use fluid modelling. 

[image]

yuvaltassa, 2022-09-20:
Mmmm I think I would still try very hard to use a DoF-less model. It would be **orders of magnitude** faster.

Not sure if this is up your alley, but the "correct" way to do this is to introduce a new plugin (see last bullet of [latest changelog](https://mujoco.readthedocs.io/en/latest/changelog.html)), which would basically replicate the code of one of the current fluid models, but make wind a function of global position.

You can start by just building MuJoCo from source and hacking away at the [fluid models](https://github.com/deepmind/mujoco/blob/main/src/engine/engine_core_smooth.c#L1495)? Wherever you see `wind` introduce your velocity field. 

As a very rough first approximation, you can forget the fluid models altogether and just use `xfrc_applied` to apply a position-dependent force-field to all the bodies.

To be completely honest, depending on what exactly you are hoping to achieve, it could very well be that MuJoCo is not the best tool for your use-case. [SOFA](https://github.com/sofa-framework/sofa) are considered quite good at biomedical and soft-body sim, though we are not SOFA experts by any means.

closed 2022-09-20 by yuvaltassa

tudorjnu, 2022-09-21:
In this case, I will initially try to adjust the `xfrc_applied` and will further build MuJoCo from source and adjust the fluid model. Perhaps I will try to import an existing fluid model from other libraries and see if they can be integrated. 

I will be on the lookout for the plugin API that is about to drop. Is there an approximate date on which it will be released? 

Unfortunately SOFA does not have good documentation and support (the website has been down for months at this time). In contrast, since the acquisition, MuJoCo has been evolving in a very fast-paced manner, with extremely well-written documentation and exceptional support which is why I chose it as a base in the first place. 

Thank you for the suggestions! I will see how it goes.

yuvaltassa, 2022-09-21:
To clarify, the plugin API is already public, merely not in any released version. If you build from source you will get it.
We intend to release first party plugins soon illustrating how to use it, but in the meantime just hack away at the code and see where you get :slightly_smiling_face: 
We are here to help.

tudorjnu, 2022-09-21:
Sweet! This helps a lot and it is very comforting! Thanks again for all the support! :)