google-deepmind/mujoco #500: How can I limit the length of a tendon in simulation?

takuyaliu, 2022-09-23:
Hi,

I'm a student and I'm trying to use MuJoCo for my robot simulation.

I'm looking for some help with the use of tendon.

Here is a model which explains my question:

In mujoco's official document, I find a xml file which is example of tendon--tendon.xml

[image]

In tendon.xml, I found that the length of the tendon is limited:

[image]

When the cylinder drops, tendon will stretch. So I try to use tendon's characteristic to model the ropes in my robot. Here is my robot:

https://youtu.be/4g4OXuEegLY

And this is part of my code and the site I defined:

[image]
[image]

You can see that the length of the tendon didn't change all the time.

I also try to use the rope model in mujoco's document, but I found that I can't specify the roe's path:

https://youtu.be/J1mmr_WmGuM

--------------------------------------------------------------------------------------------------------------------------------------------------

So I want to ask some questions about the use of tendon:

How can I limit the length of tendon ? Or my method is wrong ?

How to simulate a rope that can be linked with the robot joint ?

Thanks for reading, looking forward to your reply.

yuvaltassa, 2022-09-23:
Take a look at [this model](https://github.com/deepmind/mujoco/blob/main/test/engine/testdata/catenary.xml), it contains an example of a limited tendon

A video of this model it can be found [here](https://www.youtube.com/watch?v=I2q7D0Vda-A).

Does this help?

takuyaliu, 2022-09-24:
> Take a look at [this model](https://github.com/deepmind/mujoco/blob/main/test/engine/testdata/catenary.xml), it contains an example of a limited tendon
> 
> A video of this model it can be found [here](https://www.youtube.com/watch?v=I2q7D0Vda-A).
> 
> Does this help?

Thanks for your reply ! I've simulated this model, I think it's very helpful for my robot's simulation, I will try to implement it on my model.

But I don't understand why this model's tendon can be limited, the variables I use are similar to this model except for the hint joint and actor, but I cant't get the same effect.

Is it necessary to define the hinge joint ? Or the actuator ?

Thank you very much for your answer again ! Wish you a happy life work !

yuvaltassa, 2022-09-24:
You are using a maximum tendon length of 60cm, are you sure that is correct? Seems a bit "too round" to be precise.

I would strongly recommend adding a [tendon length sensor](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-tendonpos) whose values you can print or display in simulate (F4, I think).

This should help you debug.

Try this suggestion first and if you still fail you can attach your model here and I will take a look. If you do that, please make sure the model is loadable before attaching it (i.e. include the assets etc.)

takuyaliu, 2022-09-25:
> You are using a maximum tendon length of 60cm, are you sure that is correct? Seems a bit "too round" to be precise.
>
> I would strongly recommend adding a [tendon length [...]

Thanks for replying !

I‘ve try to reduce the length of the tendon, and I used 0.3m rope before, but I changed it to 0.22m rope, which seems to work. I uploaded the simulation video to the link below, and if you are free, you can see whether it works correctly.

https://youtu.be/3VW2LpPg5ns

If you think the simulation effect of this tendon is still abnormal, I will upload the model code here.

Thank you very much for your answer again ! Wish you a ahappy life work !