google-deepmind/mujoco #501: Accessing Joint-based or Total Body Potential/Kinetic Energy via GUI and Programmatically (Mujoco-py)

sunshineinsandiego, 2022-09-24:
Hi,

I'm a student and I'm trying to use MuJoCo for humanoid movement simulation.

I'm looking for some help with accessing potential energy stored in each joint (based on the spring reference point) as well as the total potential/kinetic energy of the entire humanoid.

I modified the [standard humanoid model](https://github.com/deepmind/mujoco/blob/main/model/humanoid/humanoid.xml) to include the energy flag per the [XML reference](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag):
```
  <option>
    <flag energy="enable"/>
  </option>
```

Also per the reference, I think field should be available in the GUI as it is stored in `mjData.energy`, but I don't know how to view that value in the GUI. Here is a screenshot with the "invalid field" error I receive when I try to use the "Watch" tool:

[image]

Additionally, I should be able to access the energy field programatically. Using the standard [humanoidstandup_v4 environment ](https://github.com/openai/gym/blob/master/gym/envs/mujoco/humanoidstandup_v4.py), my thought is that adding `data.energy.flat` to the `__get__obs` function on [line 203](https://github.com/openai/gym/blob/master/gym/envs/mujoco/humanoidstandup_v4.py#L203) should give me total kinetic and potential energy in a vector of length 2. 

If that is true, is there a way to find the components of the calculated total kinetic/potential energy based on its spring-based joint components and velocity/position based `geom` body components?

Thank you!

yuvaltassa, 2022-09-24:
To view energy in the GUI hit 'F2', it should be in the info overlay.

> Additionally, I should be able to access the energy field programatically. Using the standard [humanoidstandup_v4 environment [...]

That is correct.

> If that is true, is there a way to find the components of the calculated total kinetic/potential energy based on its spring-based joint components and velocity/position based geom body components?

Not sure what you are asking here. The code which computes energy is [here](https://github.com/deepmind/mujoco/blob/main/src/engine/engine_sensor.c#L774), I think it's pretty readable and can easily be recreated in Python if you wish to modify.

closed 2022-09-24 by yuvaltassa

sunshineinsandiego, 2022-09-25:
Great, thank you! Very helpful. Re:Q2, I was simply looking for a way to access a vector of all the joint-based energy components that are used to calculate the total energy. But from the code it looks like it's stored as a running total in `energy[0]` so if I want to access each individual joint's energy, I will need to recalculate each time.