google-deepmind/mujoco #503: IndexOutOfRangeException for importing mujoco_menagerie models into Unity

shreyasgite, 2022-09-26:
**IndexOutOfRangeException: The mesh exceeds the number of vertices per mesh allowed by Unity. (2728366986 > 65535)**

Hi,

I'm using MuJoCo with unity plugin. And trying to import the mujoco_menagerie models.
However Unity won't let me do it,  and throws out following error message:

`IndexOutOfRangeException: The mesh exceeds the number of vertices per mesh allowed by Unity. (2728366986 > 65535)
Mujoco.StlMeshParser.ParseBinary (System.Byte[] stlFileContents, UnityEngine.Vector3 scale) (at /Users/shreyas/mujoco/unity/Editor/Importer/StlMeshParser.cs:64)
Mujoco.MjImporterWithAssets.CopyMeshAndRescale (System.String sourceFilePath, System.String targetFilePath, UnityEngine.Vector3 scale) (at /Users/shreyas/mujoco/unity/Editor/Importer/MjImporterWithAssets.cs:181)
Mujoco.MjImporterWithAssets.ParseMesh (System.Xml.XmlElement parentNode) (at /Users/shreyas/mujoco/unity/Editor/Importer/MjImporterWithAssets.cs:160)
Mujoco.MjImporterWithAssets.ParseAssets (System.Xml.XmlElement parentNode) (at /Users/shreyas/mujoco/unity/Editor/Importer/MjImporterWithAssets.cs:125)
Mujoco.MjImporterWithAssets.ParseRoot (UnityEngine.GameObject parentObject, System.Xml.XmlElement parentNode) (at /Users/shreyas/mujoco/unity/Editor/Importer/MjImporterWithAssets.cs:88)
Mujoco.MjcfImporter.ImportXml (System.Xml.XmlDocument mjcfXml, System.String name) (at /Users/shreyas/mujoco/unity/Runtime/Importer/MjcfImporter.cs:100)
UnityEngine.Debug:LogException(Exception)
Mujoco.MjcfImporter:ImportXml(XmlDocument, String) (at /Users/shreyas/mujoco/unity/Runtime/Importer/MjcfImporter.cs:104)
Mujoco.MjImporterWithAssets:ImportString(String, String, String) (at /Users/shreyas/mujoco/unity/Editor/Importer/MjImporterWithAssets.cs:74)
Mujoco.MjImporterWithAssets:ImportFile(String) (at /Users/shreyas/mujoco/unity/Editor/Importer/MjImporterWithAssets.cs:59)
Mujoco.MjImporterEditorWindow:Apply() (at /Users/shreyas/mujoco/unity/Editor/Importer/MjImporterEditorWindow.cs:28)`

Is there a workaround for this? Thanks a lot!

kevinzakka, 2022-09-26:
No Menagerie model has 2.7B vertices, this looks like a Unity-specific bug.

yuvaltassa, 2022-09-26:
I'm suspecting some sort of underflow. 

@shreyasgite, can you specify which model you are trying to import?

Also, Unity version, MuJoCo version, operating system, generally [stuff required to reproduce](https://github.com/deepmind/mujoco#asking-questions).

shreyasgite, 2022-09-27:
Sorry about not giving this info before. Thanks Kevin. 

I am trying to import this into Unity to create simple robot learning sims from computer vision perspective. 

The above error happens with all the mujoco_menagerie  (Cassie, anymal, panda, unitree) models. But the import works for standard models form Mujoco library (humanoid, humanoid 100, tendon arm etc)

OS: macOS Monterey 12.6
Chip: Apple M1 Max
Unity: 2021.3.10f1
Mujoco: 2.2.2

teamclouday, 2023-01-14:
@pkrack Thank you very much! I'm able to load panda.xml by following your steps.

Norite-dev, 2023-05-12:
@pkrack thank you, I've done everything you've said, but still something goes wrong. 
OS: MacOS Ventura 13.3.1
CPU: Intel Core i5
Unity: 2021.3.23f1
MuJoCo: 2.3.5

When I press "Play" there is an error:

Error:
NullReferenceException: Failed to create Mujoco runtime.
Mujoco.MjScene.StepScene () (at /Users/akvelon/Desktop/mujoco-main/unity/Runtime/Components/MjScene.cs:345)
Mujoco.MjScene.FixedUpdate () (at /Users/akvelon/Desktop/mujoco-main/unity/Runtime/Components/MjScene.cs:98)

https://images2.imgbox.com/a8/cf/ZUR6m4fO_o.png

could you please suggest something? 
Thanks

closed 2023-11-22 by kevinzakka