google-deepmind/mujoco #505: Regarding the new boxQP implementation

nrkumar93, 2022-09-27:
Glad to see the new [boxQP](https://github.com/deepmind/mujoco/blob/1e2a9a53bcca2e2bf6758b350db32346d132098d/include/mujoco/mujoco.h#L1036) implementation and looking forward to the next release. 

Just want to ask if it 
1. implements the Projected Newton Method as it is not explicitly stated anywhere yet.
2. is ready to use or is a work in progress?

Thanks

yuvaltassa, 2022-09-27:
1. As stated in the changelog, it implements the algorithm described in the appendix of [this paper](https://doi.org/10.1109/ICRA.2014.6907001). Yes, it is a projected Newton method.

2. It is ready to use, but there is a an upcoming small modification. Currently it assumes you are giving it a symmetric Hessian. If you give it an asymmetric one it will fail to converge. After the upcoming change it only reads from the lower triangle which will effectively symmetrise the Hessian, making the above issue moot.

closed 2022-09-27 by yuvaltassa

yuvaltassa, 2022-09-27:
a6b338db6481ff6d94ea62a11eecd90962d8903d was jus pushed with the aforementioned improvement.