google-deepmind/mujoco #506: Setting up MuJoCo (Python Bindings)

SerhiiBahdas, 2022-09-27:
**Setting up MuJoCo (Python Bindings)**
Hello,

I'm a PhD student in Neuroscience and I'm trying to use MuJoCo to create a multi-DOF human arm model.

I'm looking for some help with using Python Bindings. After importing MuJoCo `import mujoco` and trying yo use its modules, for example, `MjModel`, I always get an error shown on the screenshots below. I can't wait to start coding, but this minor issue prevents me from doing so. 

Example from Spyder (using it through Anaconda):
[image]

Example from Terminal:
[image]

Thanks a lot, 
Serhii

saran-t, 2022-09-27:
A bit of a guess here, but I think your Anaconda installation is using Intel binaries rather than Apple Silicon. Are you using an Apple Silicon machine?

SerhiiBahdas, 2022-09-28:
Hi Saran, 

Thanks a lot for getting back to me this fast! Yes, I am using an M1 Mac, and I overlooked that I installed the Intel-based Anaconda. I wiped it out, reinstalled Anaconda, and everything works now! Thank you! 🙌

[image]

closed 2022-09-28 by SerhiiBahdas