google-deepmind/mujoco #513: Is there a way to make tendons collide with one another?

traplotz, 2022-10-06:
Hi,
I'm a student and I'm trying to make a stretchy rope by combining a series of geom spheres with short tendons in between.
(I'm doing this because I've tried using rope with main+stretch joints, but I found it kind of unstable and I wasn't happy with the result.)

The stretchy-ness of the rope is working great, but the trouble is that the rope does not get wound as a rope is supposed to.
As shown in the figure below, the pitch of the coil changes with tension, because the contact is working only among the geom spheres and not among the tendons.
[image]

So, I want to make my tendons collide with one another just like the geoms do.
I know it's a weird thing to ask, but to me it kind of seems like there is some sort of contact between tendons in the [Tendon-driven leg](https://youtu.be/mfAst_GB8Sk) example. Or maybe it's just transparent geoms.

I would be much obliged with any piece of advice.

yuvaltassa, 2022-10-06:
This was asked and answered [2 days ago](511). The answer is unfortunately no, tendons cannot collide other than with wrapping geoms (which is not what you want, in your case).

As a temporary solution you can try to use capsules rather than spheres?

I can add that improving soft bodies generally in MuJoCo, and rope-like 1D soft bodies in particular, is something we consider high priority and are actively working on. When these are released we'll also ping you here in this issue.

Cheers

closed 2022-10-06 by yuvaltassa

quagla, 2022-10-06:
Hi traplotz, we will soon release a new rope/cable which is more stable than the current one. We will let you know once this is completed, it would be great if you could test it. The initial release won't be stretchable, since that would require updating the geometries for collisions, which is not supported yet. But it would be easy to add a stretchable joint as currently done for the composite rope (I could give you code for doing this).

traplotz, 2022-10-07:
@youvaltassa
I tried using capsules and it kind of does the job. The rope touches itself much better than it used to with spheres, and the pitch stays constant with different tensions.

The rope has become quite unstable, though. When I give the rope some tension and wait long enough, the capsules start to vibrate more and more wildly, and suddenly the runtime ends and starts over.
I'm not sure why but it seems like the global kinetic energy of the rope is transferred to each capsule and causes some sort of resonance.
I am testing with different site pos and tendon damping, etc, but the only remedy that seems to work is shorter timestep.

traplotz, 2022-10-07:
@quagla 
It would be an honor to participate, but I'm not sure if I am qualified for the testing. 
I am just a beginner and I dont' even know how to use the API yet.
If some inept beginner-level comments would be fine, please let me test the code.
Else, thanks for asking though.

quagla, 2022-10-18:
Hi I'm glad to inform you that the new cable is available. Please have a look at https://mujoco.readthedocs.io/en/latest/changelog.html where there are 3 examples. I am really curious to know if you find this cable more stable than the rope. 

Note that this cable is unstretchable, but you can add stretching either in the code (which sounds like you would rather avoid) or directly in the XML. If you load and save one of the 3 examples, you will see how the Composite is expanded in each single element that composes the rope. You could add one slider joint per element in order to obtain your stretch.