google-deepmind/mujoco #518: Python bindings named access does not return id

cube1324, 2022-10-07:
In the Python Bindings Documentation this is an example for named access:

> _m.geom('gizmo').id_ is the same number as returned by _mujoco.mj_name2id(m, mujoco.mjtObj.mjOBJ_GEOM, 'gizmo')_

However when I try to replicate this, it does not work

```python
import mujoco

XML=r"""
<mujoco>
  <worldbody>
    <body>
      <geom type="box" size="1 1 1" name="gizmo"/>
    </body>
  </worldbody>
</mujoco>
"""

model = mujoco.MjModel.from_xml_string(XML)
data = mujoco.MjData(model)

id1 = mujoco.mj_name2id(model, mujoco.mjtObj.mjOBJ_GEOM, 'gizmo') # Works
id2 = model.geom("gizmo").id # Fails with AttributeError: 'mujoco._structs._MjModelGeomViews' object has no attribute 'id'
```

Im using Mujoco Version 2.2.2 and Windows 10

yuvaltassa, 2022-10-07:
Hi!

MuJoCo "latest" documetation describes the code at head, not the last release. You can click the "v:latest" button on the bottom left to see the 2.2.2 documentation which you'll see does not include this feature.

 This feature will be available at the next release, or your can build MuJoCo from source yourself.

closed 2022-10-07 by yuvaltassa

yuvaltassa, 2022-10-18:
Fixed by 57b0b36699e26fa1bda20e58a059d7ac7e439b3a
This feature is now available in pypi wheels.