google-deepmind/mujoco #520: Simulating elastic tendon with tension-only forces

kgaranger, 2022-10-09:
Is is possible for tendons to behave like elastic strings, that is to produce an elastic force only in tension, when their length is greater that their rest length?

I am aware of the rope element but I am not interested in the tendon's state itself not do I wish to assign it a mass so that would not work. So far I've been able to come close to the desired behavior by limited the range of a tendon and using the solver parameters for the limit so that they correspond to the desired stiffness, but it seems overly hacky and results in a much slower simulation for a large number of tendons.

The following shows an example of how the desired behavior is achieved:

```
<mujoco model="">
  <option timestep="1e-02"/>
  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1="0.1 0.2 0.3" rgb2="0.2 0.3 0.4" width="512" height="512"/>
    <material name="grid"  texture="grid" texrepeat="1 1" texuniform="true" reflectance="0.2"/>
  </asset>
  <default>
      <geom type="sphere" size="0.1" rgba="0.4 0.6 0.4 1"/>
      <default class="free">
          <geom rgba="0.6 0.2 0.6 1"/>
          <joint type="free"/>
      </default>
  </default>

  <worldbody>
      <!-- Tension only elastic forces -->
      <body pos="-0.6 0.6 0.3">
          <site name="s2"/>
          <geom/>
      </body>
      <body pos="0.1 0.6 0.1" childclass="free">
          <joint/>
          <site name="s3"/>
          <geom/>
      </body>

      <!-- Tension and compression elastic forces -->
      <body pos="-0.6 -0.4 0.3">
          <site name="s0"/>
          <geom/>
      </body>
      <body pos="0.1 -0.4 0.1" childclass="free">
          <joint/>
          <site name="s1"/>
          <geom/>
      </body>
      
      <!-- Following bodies are to ensure that resulting stiffness are equal in both cases -->
      <body pos="0.6 -0.4 1.0981">
          <site name="s4"/>
          <geom/>
      </body>
      <body pos="0.6 -0.4 0.3" childclass="free">
          <joint/>
          <site name="s5"/>
          <geom mass="1"/>
      </body>
      
      <body pos="0.6 0.6 1.0981">
          <site name="s6"/>
          <geom/>
      </body>
      <body pos="0.6 0.6 0.3" childclass="free">
          <joint/>
          <site name="s7"/>
          <geom mass="1"/>
      </body>

    <geom type="plane" size="0 0 0.05" material="grid" rgba="1 1 1 1"/>
    <light mode="trackcom" pos="0 0.5 6" diffuse="0.8 0.8 0.8" specular="0.3 0.3 0.3"/>
  </worldbody>
  <tendon>
    <spatial stiffness="100" springlength="0.5">
      <site site="s0"/>
      <site site="s1"/>
    </spatial>
    <spatial limited="true" springlength="0.5" range="0 0.5" solimplimit="0.001 0.001 0.001 0.5 1" solreflimit="-100 -0" >
      <site site="s2"/>
      <site site="s3"/>
    </spatial>

    <spatial springlength="0.7" stiffness="100">
      <site site="s4"/>
      <site site="s5"/>
    </spatial>
    <spatial springlength="0.7" limited="true" range="0 0.7" solimplimit="0.001 0.001 0.001 0.5 1" solreflimit="-100 -0" >
      <site site="s6"/>
      <site site="s7"/>
    </spatial>
</tendon>
</mujoco>

```

By the way, how come that the stiffness in `solreflimit` does not have to be scaled by the inverse of the impedance to result in the same effective tendon stiffness, based on the formula for the dynamics in constraint space given in the [Modeling > Solver parameters](https://mujoco.readthedocs.io/en/latest/modeling.html#solver-parameters) of the documentation?

yuvaltassa, 2022-10-09:
Your solution is a valid one.

Did you really experience this?
> results in a much slower simulation for a large number of tendons.

I would imagine the slowdown to be rather minimal.

That said, this is a valid feature request and not very difficult to implement. I am imagining an additional attribute, perhaps `<spatial springside>` which if 0 (the default ) is two-sided, if negative produces only tension and if positive produces only compression. This attribute could work for `<joint>` as well as tendons.

Can you confirm that this is the feature you are asking for? Can you think of a better implementation?

kgaranger, 2022-10-09:
Thank you for response.

> Did you really experience this?
> 
> > results in a much slower simulation for a large number of tendons.

Yes, the mjcf file below provides a comparison for a more complex structure that I am working with. Switching from using the stiffness attribute to the solreflimit attribute can be done by commenting and uncommenting the relevant default elements at the beginning. The model is pretty stiff so runs slowly in any case, by I get a 4x slower computation time when I use solreflimit.
I also observe a different behavior which I cannot explain. The model using solreflimit seems to be much less stiff than the other one. It could be due to the difference in the dynamics of both models induced by compressive forces in one and not in the other, but I doubt that this is the case since the model is designed such that tendons should only be in tension.

> That said, this is a valid feature request and not very difficult to implement. I am imagining an additional attribute, perhaps `<spatial springside>` which if 0 (the default ) is two-sided, if [...]

I confirm that this is the feature I am asking for and I was imagining the same mechanism. I can contribute on that. I will probably implement it for my work given the issues I have with my current solution and I can then work on a pull request. Are there examples of existing attributes where the value needs to be distinguished between negative, positive, or zero?

<mujoco model="Truncated octahedron tensegrity">
  <option timestep="1.0000000000000001e-05"/>
  <default>
    <default class="/">
      <geom solref="0.0001 1"/>
      <equality solref="0.0001 1"/>
      <default class="bar">
        <joint type="slide" axis="0 0 1" limited="true" stiffness="862372.04334828781" range="0.52212820504653834 0.63815669505688022" springref="0.58014245005170928" damping="94.766158609701947"/>
        <geom type="capsule" size="0.0040000000000000001" rgba="0.40000000000000002 0.40000000000000002 0.40000000000000002 1" mass="0.0022964397864887218" fromto="0 0 0 0 0 0.29007122502585464"/>
        <site size="9.9999999999999995e-07"/>
      </default>
      
      
      
      <default class="cable">
        <tendon width="0.002" rgba="0.20000000000000001 0.5 0.80000000000000004 0.90000000000000002" stiffness="274637.74945477257" damping="0"/>
      </default>
    </default>
  </default>
  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1="0.10000000000000001 0.20000000000000001 0.29999999999999999" rgb2="0.20000000000000001 0.29999999999999999 0.40000000000000002" width="512" height="512"/>
    <material name="grid" class="/" texture="grid" texrepeat="1 1" texuniform="true" reflectance="0.20000000000000001"/>
  </asset>
  <worldbody>
    <body name="member_0" pos="0.15976599999999999 -0.316992 0.30159200000000003" zaxis="-0.31953199999999998 0.34239199999999997 0.34239199999999997">
      <joint name="member_0_joint" class="/" type="free"/>
      <body name="member_0_node_0">
        <geom name="member_0_mass_0" class="bar"/>
        <site name="member_0_site_0" class="bar"/>
        <body name="member_0_node_1" pos="0 0 0.58014245005170928" zaxis="0 0 -1">
          <joint name="member_0_spring" class="bar" ref="0.58014245005170928"/>
          <geom name="member_0_mass_1" class="bar"/>
          <site name="member_0_site_1" class="bar"/>
        </body>
      </body>
    </body>
    <body name="member_1" pos="-0.025399999999999999 -0.316992 0.16722600000000001" zaxis="0.34239199999999997 0.34239199999999997 0.31953200000000004">
      <joint name="member_1_joint" class="/" type="free"/>
      <body name="member_1_node_0">
        <geom name="member_1_mass_0" class="bar"/>
        <site name="member_1_site_0" class="bar"/>
        <body name="member_1_node_1" pos="0 0 0.58014245005170928" zaxis="0 0 -1">
          <joint name="member_1_spring" class="bar" ref="0.58014245005170928"/>
          <geom name="member_1_mass_1" class="bar"/>
          <site name="member_1_site_1" class="bar"/>
        </body>
      </body>
    </body>
    <body name="member_2" pos="0.15976599999999999 0.025399999999999999 0.010000000000000009" zaxis="-0.31953199999999998 -0.34239199999999997 0.34239199999999997">
      <joint name="member_2_joint" class="/" type="free"/>
      <body name="member_2_node_0">
        <geom name="member_2_mass_0" class="bar"/>
        <site name="member_2_site_0" class="bar"/>
        <body name="member_2_node_1" pos="0 0 0.58014245005170928" zaxis="0 0 -1">
          <joint name="member_2_spring" class="bar" ref="0.58014245005170928"/>
          <geom name="member_2_mass_1" class="bar"/>
          <site name="member_2_site_1" class="bar"/>
[... 238 lines cut ...]
    </spatial>
    <spatial name="member_43" class="cable" springlength="0.22808040282321493">
      <site site="member_7_site_0"/>
      <site site="member_10_site_1"/>
    </spatial>
    <spatial name="member_44" class="cable" springlength="0.22808040282321493">
      <site site="member_5_site_0"/>
      <site site="member_1_site_0"/>
    </spatial>
    <spatial name="member_45" class="cable" springlength="0.22808040282321493">
      <site site="member_10_site_0"/>
      <site site="member_3_site_0"/>
    </spatial>
    <spatial name="member_46" class="cable" springlength="0.22808040282321493">
      <site site="member_4_site_0"/>
      <site site="member_11_site_0"/>
    </spatial>
    <spatial name="member_47" class="cable" springlength="0.22808040282321493">
      <site site="member_2_site_0"/>
      <site site="member_9_site_0"/>
    </spatial>
  </tendon>
  <sensor>
  </sensor>
</mujoco>

yuvaltassa, 2022-10-09:
Ooooh a tenesegrity structure! I would love to see this when it's ready.

Giving this feature a bit more thought, I think maybe this is not the best parametrisation. For example if I want to do this for a joint, the one-sidedness is a limitation. For example I can imagine a slider joint with limits that are implemented as springs rather than constraints. Cheaper and easier to analyse. Maybe what we want is a true/false attribute that means "implement the specified range limits with passive springs, using the specified stiffness", (`springlimits` perhaps?). If this attribute is true then setting `springref` is an error, these attributes should be exclusive. Of course the question now becomes, would you ever want _both_ constraint limits and some sort of one-sided spring?

I think it's worth the effort to think this through beforehand, especially as the implementation will be fairly straightforward either way. So thinking carefully about future use cases and gotchas is actually very valuable.

Thanks for the offer to work on this! Much appreciated.

I'll take a look at your other question later this week, I hope.

kgaranger, 2022-10-10:
> Ooooh a tensegrity structure! I would love to see this when it's ready.

I will share when it is.

> Giving this feature a bit more thought, I think maybe this is not the best parametrisation. For example if I want to do this for a joint, the one-sidedness is a limitation. I can imagine a slider joint with limits that are implemented as springs rather than constraints. Cheaper and easier to analyse. Maybe what we want is a true/false attribute that means "implement the specified range limits with passive springs, using the specified stiffness", (`springlimits` perhaps?). If this attribute is true then setting `springref` is an error, these attributes should be exclusive. Of course the question now becomes, would you ever want _both_ constraint limits and some sort of one-sided spring?

Good points. I can possibly see cases where a one-sided spring with constraints limits is needed. But I can also imagine how a work-around would always be possible. For example, for a tendon, by instead using two tendons in parallel, one for the constraint limits with zero stiffness and one for the one-sided spring. This option for a joint may be less desirable because it would require using equality constraints, although if the joint is a slider one, one could use a joint and a tendon in parallel. So there remains the case of wanting some one-sided hinge joint with constraint limits, which could probably always be dealt with by using `geom` objects to enforce limits.

yuvaltassa, 2022-10-11:
Hi @kgaranger,

We've been discussing internally, here are some thoughts. There are 3 separate things which make sense in the context of your feature request:

1. A very general passive force plugin that would allow the user to inject any kind of spring/damper force into joints and tendons. This would also be a good example of how to write a plugin and could be generally useful for advanced users.
2. A mechanism that would replace any constraint with a spring-damper. This would work the same way for limits and for contacts. Frictional contact and joint friction are a bit different since there is no constraint violation, but one could replace them with a fixed force opposing the velocity. This option is conceptually simple but might be a bit involved in terms of the code structure.
3. Extending the existing native springs for joints and tendons. There are two simple extensions which make sense:
  * Adding a `springrange` attribute (exclusive with `springref`), which would create a deadzone where the spring is inactive. If you set `springrange` equal to `range`, the spring acts in the same way as limits, but does not replace them and you can have both if you like with different ranges.
     - One could also make `springref` an optionally 2-value attribute such that when there are two values it is interpreted as a range.   
  * Extending stiffness with higher order derivatives such that e.g. `stiffness="a b c"` means `f(x) = ax + bx^2/2*sign(x) + cx^3/6` etc.
  
What are your thoughts on these options? Specifically:
1. Do you want higher-order springs (apart from the deadzone)?  
2. Would you use a general purpose passive force plugin if we provided one? 
3. Do you prefer the proposed new `springrange` attribute or do you think `springref` should take 2 values?

kgaranger, 2022-10-14:
Thank you for putting some thought into this @yuvaltassa. I like both options 1 and 3.
To answer your questions specifically:
1. I don't think I will need higher-order springs.
2. I probably would use a general purpose passive force plugin especially if option 3 is not implemented. If option 3 was implemented I don't think that there would be an immediate need for such plugin. Also if there was a plugin like that, it may desirable to require users to also implement a function to give the corresponding potential energy, in case the computation of a model's energy is needed.
3. I think `springrange` is better.

closed 2022-10-28 by copybara-service[bot]

yuvaltassa, 2022-10-28:
Hi @kgaranger,

I just submitted a fix for this, basically supporting exactly what you asked for in the OP (though using `springlength` rather than `springrange`). You'll need to build from source to get this feature (or wait for the next release). Load the example model in `simulate` and let me know if something is not obvious. It is still not ideal of course, for example you probably want damping to be turned off when the spring is in the deadzone, but I feel that for any more advanced passive-force features should be implemented as a plugin.

Let me know how you get on and send us that tensegrity model when it's ready! Would love to include it in the [Menagerie](https://github.com/deepmind/mujoco_menagerie).

kgaranger, 2022-11-11:
Thank you! It works as expected. I will let you know when the model is ready.

wangsd01, 2023-01-12:
I understand new springlength parameter with two values will create a dead zone. I was thinking if I combine an actuator with the tendon, e.g. release or retract the tendon, the dead zone will change accordingly? Which actuator should I use in this case? Thanks!

commit 893942a7 (2022-10-28): Allow tendon `springlength` attribute to take two values between which the spring produces no force (a deadband).

If given two non-decreasing values, `springlength` specifies a deadband for spring stiffness, inside of which the force is 0 and outside of which force behaves like a regular spring, with the setpoint corresponding to the nearest value. This can be used to create tendons whose limits are enforced by springs rather than constraints, which are cheaper and easier to analyse. See test/engine/testdata/tendon_springlength.xml example model.

Spring-limited spatial tendons whose lower ra [... cut]