google-deepmind/mujoco #532: Migration from mujoco-py to mujoco

aadharna, 2022-10-15:
Hi,

I'm a student and I'm trying to use MuJoCo for RL research.

I'm looking for some help with finding a migration guide for taking [software written for mujoco-py](https://github.com/openai/mujoco-worldgen/blob/master/requirements.txt) to proper mujoco now that there are official bindings. If I just replace the (inherited) dependence on `mujoco-py<1.50.2,>=1.50.1` with mujoco in the requirements.py file is there a chance it just works out of the box?

I did a quick search through the discussion pages and other open issues, but didn't see anything on this. I did find a migration guide from deepmind_mc to mujoco, but that wasn't what I was looking for.

kevinzakka, 2022-10-15:
There's a [mini guide](https://mujoco.readthedocs.io/en/latest/python.html#migration-notes-for-mujoco-py) in the docs, hopefully that's helpful.

aadharna, 2022-10-15:
Wow. I don't know how I missed that. Thanks. I'll give that a read later today! Thank.

For now, I'll close this.

closed 2022-10-15 by aadharna