google-deepmind/mujoco #534: Hybrid dynamics with contact

lieskjur, 2022-10-18:
Hi, I would like to ask how best problems of hybrid dynamics, arrising from contact, can be solved using mujoco.

As an example, let's say that I am controlling the "cheetah" from the dm_control suite and I want to position it's head while it is standing on the ground. One possible approach would be to prescibe its head some trajectory with continuous second derivatives and solve a problem that would be classified as hybrid dynamics (according to Featherstone).

Precisely, I would have some demands on the accelerations of `root_x`, `root_z` and `root_y` while the accelerations of other joints could be optimized along with actuators' inputs. In this case it would be a generalized problem of hybrid dynamics with over-actuation but similar exactly-actuated problems (number of desired accelerations equals the number of actuators) could also be formulated.

In rigid body dynamics without contact the exactly-actuated problem has an exact solution while the over-actuated problem can be solved using QP with a cost function such as u'Ru and equality constraints in the of EoM.

This leads me to think that an approach to finding a solution to exactly-actuated hybrid dynamics with contact should be compatible with the way mujoco handles contact (soft penalty).

yuvaltassa, 2022-10-18:
This comment is rather confusing.

First, what is "hybrid dynamics"? I think perhaps you mean the approach whereby contact is represented as bilateral rather than unilateral constraints? Maybe you mean something else? Please clarify.

Second, you open your comment by saying that you would like to ask a question, but then there is no question. Do you have a question?

lieskjur, 2022-10-18:
In [Featherstone's book](https://link.springer.com/book/10.1007/978-1-4899-7560-7) the hybrid dynamics' problem (of an exactly actuated mechanisms without kinematic loops) is loosely defined as finding m applied forces (in joint space) and n-m joint accelerations for given values of the remaining n-m forces and m accelerations (n is the number of DoF). So in effect it is a mix of the forward and inverse dynamics problem.

A very similar problem to this are the inverse dynamics of a system with any form of constraints (kinematic loops). Where we can set only a select number of desired joint accelerations as other must be calculated so that the constraints are satisfied.

I wanted to ask if/how mujoco can be used to solve this type of problems when the constraints are the result of contact.

I hope I explained it a bit better now.

yuvaltassa, 2022-10-19:
I see, thanks for the explanation. 

If I understand the setup correctly, I don't think it is directly doable in MuJoCo, except perhaps by some optimisation procedure.

Note however that inverse dynamics is well defined in MuJoCo in the presence of contacts. That is to say that given {`qpos`, `qvel`, `qacc`}, a call to `mj_inverse` will compute not just the applied forces `qfrc_inverse`, but also all contact/constraint forces in `efc_force` (and its projection – `qfrc_constraint`).

Take a look at our LQR colab (item 2 in the 2.3.0 [changelog](https://mujoco.readthedocs.io/en/latest/changelog.html)), where this property is explicitly made use of.

lieskjur, 2022-11-18:
Sorry for the long delay, I didn't want to reply before I had time to read through the relevant documentation properly.

If I understand it correctly, mujoco's soft constraint model makes all `qacc` feasible along with other benefits. An example of a downside in comparison to rigid contact models is then that a "magic" force would have to hold up the humaniod in the colab you referenced.

From what I read in the [paper](https://ieeexplore.ieee.org/document/8462904), Goal-Directed Dynamics should be capable of solving a bigger family of problems, including the one that I described, in the future which I was glad to find out. For now I'll go back to exporing my options, thank you for your time.