google-deepmind/mujoco #538: Loose initial config/status of rope

zxwang96, 2022-10-18:
Hi,

I'm trying to use MuJoCo for simulate the scenario that one end of a rope holding by a UR5 robot is connected to a fixed anchor point (the colored ball). The position of the anchor point is not at the end of the fully stretched rope.

I used a equality weld to build the connection as the description said it will omit the DoF and relative position of body 1 & 2. However, as the video shows the rope is not really attached to the anchor point at the very beginning, in contrast, the rope is attracted to the anchor (but can't really reach to the anchor's position).

Here is the related of snippet of the full XML. 

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
       ... ...
                                    <body name="EE" pos="-0.04 0 0" euler="0 0 0">
                                        <body name="B0" pos="-0.03 0 0" euler="0 3.14 0">
                                            <!-- <freejoint/> -->
                                            <composite type="rope" count="96 1 1" spacing="0.04" offset="0 0 2">
                                                <joint kind="main" damping="0.005"/>
                                                <geom type="capsule" size=".01 .015" rgba=".8 .2 .1 1" mass="0.005"/>
                                            </composite>
                                        </body>
                                    </body>
       ... ...
       <body name="anchor_point" pos="-3 0 1">
            <geom type="sphere" size="0.03" rgba="0 1 0 1"/>
        </body>
  </worldbody>

  <equality>
      <weld body1="anchor_point" body2="B95" relpose="0 0 0 1 0 0 0" solref="-1e4 -1e3"/>
  </equality>
  ...
</mujoco>
```

</details>

Here is a video illustrating my question:

https://user-images.githubusercontent.com/28090369/196514349-246d283d-8e30-4a07-9c58-bfe1a5272636.mp4

quagla, 2022-10-18:
Hi, the XML looks to be different from the video? But I can give you a tip for debugging the weld: if you insert `<statistic meansize="1"/>` in the XML file, then in the **Rendering** panel you can select **Constraint** and you will see a red and a blue sphere: those correspond to the anchor point in body1 and body2. The constraint solver is trying to match the two positions (and orientations). Maybe the spheres are not where you would like them to be and you have to change `relpose` or `anchor`?

zxwang96, 2022-10-18:
> Hi, the XML looks to be different from the video? But I can give you a tip for debugging the weld: if you insert `<statistic meansize="1"/>` in the XML file, then in the **Rendering** panel you can [...]

Hi! Thanks for  replay, please refer this full XML:

```ruby
<mujoco model="ur5">
    <compiler angle="radian" meshdir="meshes" balanceinertia="true"/>

    <default>
        <light castshadow="false" diffuse="1 1 1" />
        <camera fovy="60" />
    </default>

    <custom>
        <numeric name="START_ANGLES" data="0 -.67 -.67 0 0 0"/>
        <numeric name="N_GRIPPER_JOINTS" data="0" />
        <text name="google_id" data="1SjWRUl-D1FZ5fB2cy4jF4X9wTsQ5LWzo"/>
    </custom>

    <asset>
        <!-- arm models -->
        <mesh file="link0.stl"/>
        <mesh file="link1.stl"/>
        <mesh file="link1_cap.stl" />
        <mesh file="link1_connector.stl" />
        <mesh file="link2.stl" />
        <mesh file="link2_tube.stl" />
        <mesh file="link2_cap.stl" />
        <mesh file="link2_connector.stl" />
        <mesh file="link3a.stl" />
        <mesh file="link3a_connector.stl" />
        <mesh file="link3_tube.stl" />
        <mesh file="link3_cap.stl" />
        <mesh file="link3b.stl" />
        <mesh file="link3b_connector.stl" />
        <mesh file="link3b_connector2.stl"/>
        <mesh file="link4.stl" />
        <mesh file="link4_cap.stl" />
        <mesh file="link4_connector.stl" />
        <mesh file="link5.stl" />
        <mesh file="link5_cap.stl" />
        <mesh file="link5_connector.stl" />
        <mesh file="link6.stl" />
        <mesh file="link6_connector.stl" />

        <texture name="floor_tile" file="meshes/floor_tile.png" width="3000" height="3000" type="2d"/>
        <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
        <material name="tile" texture="floor_tile" specular="0" shininess="0" reflectance="0" emission="1" />
        <material name="floor" texture="texplane" specular="0" shininess="0" reflectance="0" emission="1" />
        <material name="silver" specular="1" shininess="0.5" reflectance="1" emission="0.25" rgba="0.82 0.82 0.82 1"/>
        <material name="grey" specular="1" shininess="0.5" reflectance="1" emission="0.1" rgba="0.5 0.5 0.5 1"/>
        <material name="blue" specular="1" shininess="0.5" reflectance="1" emission="0" rgba="0.678 0.847 0.902 1"/>
        <material name="black" specular="1" shininess="0.5" reflectance="1" emission="0" rgba="0.2 0.2 0.2 1"/>
    </asset>

    <worldbody>
        <!-- <geom name="floor_grey_tile" pos="0 0 -1" size="0.75 0.75 0.05" type="plane" material="tile"/> -->

        <!-- <body name="hand" pos="0 0 -0.15" mocap="true">
            <geom type="box" size=".002 .004 .006" rgba="0 .9 0 .5" contype="2" conaffinity="2"/>
        </body> -->

        <!-- <body name="target" pos="0 0 -0.15" mocap="true">
            <geom name="target" type="sphere" size="0.05" rgba=".9 0 0 .5" contype="4" conaffinity="4"/>
        </body>

        <body name="target_orientation" pos="0 0 -0.15" mocap="true">
            <geom type="box" size=".02 .04 .06" rgba=".9 0 0 .5" contype="8" conaffinity="8"/>
        </body> -->

[... 86 lines cut ...]
            </body>
        </body>

        
        <body name="anchor_point" pos="-3 0 1">
            <geom type="sphere" size="0.03" rgba="0 1 0 1"/>
        </body>

    </worldbody>
    
    <equality>
        <weld body1="anchor_point" body2="B95" relpose="0 0 0 1 0 0 0" solref="-1e4 -1e3"/>
        <!-- <weld body1="anchor_point" body2="B95" anchor="0 0 0" solref="-1e4 -1e3"/> -->
    </equality>

    <actuator>
        <motor name="joint0_motor" joint="joint0"/>
        <motor name="joint1_motor" joint="joint1"/>
        <motor name="joint2_motor" joint="joint2"/>
        <motor name="joint3_motor" joint="joint3"/>
        <motor name="joint4_motor" joint="joint4"/>
        <motor name="joint5_motor" joint="joint5"/>
    </actuator>
</mujoco>
```

zxwang96, 2022-10-19:
Hi @yuvaltassa,

I found your [answer](https://github.com/deepmind/mujoco/issues/263#issuecomment-1128866926) to issue #263 is quite related to this problem. Could you help to investigate my problem? Thank you!

yuvaltassa, 2022-10-19:
Can you please attach a loadable model? (zip file with all the assets)

zxwang96, 2022-10-19:
> Can you please attach a loadable model? (zip file with all the assets)

Thanks, please check the attached zipped file. Running 
```
python main_sim.py 
```
can start the program.
[full.zip](https://github.com/deepmind/mujoco/files/9817240/full.zip)

quagla, 2022-10-19:
Hi thanks for the file, do I understand correctly that what you want is a cable that hangs from the robot to your anchor point? Then my suggestion is to use the new cable feature that is available since yesterday's release that allows you to precisely specify the vertex coordinates. This should be much better than a cable that snaps into place at the start, dragging the robot down (unless this is what you want). Please see https://github.com/deepmind/mujoco/blob/main/model/plugin/belt.xml for an example of a cable geometry specified via vertex positions.

zxwang96, 2022-10-19:
Hi Alessio, yes that's what I want and the vertex coordinates feature is really helpful. I have a question: 
Though I can set the initial state of the rope by specifying the vertex coordinates, it takes time to calculate them by hand ( the ). Is there an affordable way to generate a series of rope coordinates given only the starting and ending point? Maybe the curve option for cable is the better option?

quagla, 2022-10-19:
if it's a cable hanging from the endpoints, the shape will be a Catenary https://en.wikipedia.org/wiki/Catenary which is a simple equation that you can use to generate points. Currently we have cos and sin as predefined options, but no cosh so you would have to generate the points yourself and then paste them in the XML.

Otherwise you can play with other shapes that are less different from the equilibrium configuration and therefore have less effect on the robot at the start.

zxwang96, 2022-10-19:
Ahhh, catenary is definitely the choice I think. Thanks!
And I come up with a potential solution: Initialize the rope/cable like before without the vertex option but add a joint (a x-y-z slider or similar) to the anchor point. After the anchor point has been moved to the designated position, change the joint type or simply remove the joint so that the anchor point body is welded to <worldbody>. Do you think this a viable solution? If it is could you direct me to the joint type change/cancel function link?

zxwang96, 2022-10-20:
Thanks. I'd like to change the joint type since after reaching to the target position, the rope will be actuated by the UR5 only, and I don't want the end welded to the anchor point move at all. So I hope to remove the slider actuator. Is that possible with our API? Thanks!

quagla, 2022-10-20:
No it's not possible but if you use a position actuator, when you stop changing its value it should remain at a fixed position (try to run https://github.com/deepmind/mujoco/blob/main/model/plugin/coil.xml and play with the slider).

zxwang96, 2022-10-20:
Thank you so much for the instruction, I'll close this issue.