google-deepmind/mujoco #539: MJMODEL.txt not found

pab47, 2022-10-18:
**How to submit a good bug report**

- Are you sure this is a bug? If not, consider using the "Ask for help" or "Feature request" templates.

- Use a clear and descriptive title.

- Make it easy to reproduce the problem.

- Include a ***minimal*** model that demonstrates the problem. If the model is small, include it as inline XML. If it requires binary assets, attach it as a zip file.

- Include a screenshot or video, if relevant.

- Include the following context:
  - Operating system.
  - MuJoCo version (and if the bug is new, the version where it used to work).
  - For Python issues, what bindings are you using (i.e `mujoco`, `dm_control`, `mujoco-py`)?

kevinzakka, 2022-10-31:
Is this still an issue? FYI it's working fine for me on 2.2.3.

pab47, 2022-11-13:
Hello Kevin, I updated to 2.3.0 on Mac. On loading the model in simulate app and pressing save xml does NOT generate the MJMODEL.txt anywhere on my computer (I did a search by date).

Are you using a mac too? If yes, where is the MJMODEL.txt generated for you?

yuvaltassa, 2022-11-13:
You need to look inside the package. Right click on the simulate app and select "show package contents", it's in there somewhere, in the resources directory, IIRC.

We can and should fix this, thanks for the reminder. In the meantime find it there.

closed 2023-01-16 by kevinzakka