google-deepmind/mujoco #541: time precision issues with mjtNum

DavidPL1, 2022-10-20:
OS: Ubuntu 20.04
MuJoCo version: 2.2.2

Using the mjtNum type for time and time arithmetics allows precision errors to occur occasionally.

Here is a minimal code example:
```
int main(int argc, char **argv)
{
	mjtNum step(0.001);    //step size
	mjtNum t0(0.0);        //starting current time
	mjtNum t1(0.0);        //starting buffered time
	float t_cond(0.02);    //time difference condition

	for (int i = 0; i < 1000; i++) {

		if (t0 - t1 != t_cond && (uint)(1000*t0) - (uint)(1000*t1) == (uint)(1000*t_cond)) {
			std::printf("Precision error occurred! Evaluated: %f != %f == %f - %f\n", t_cond, t0 - t1, t0, t1);
		}

		if ((uint)(1000*t0) - (uint)(1000*t1) == (uint)(1000*t_cond)) { // this check works
			t1 = t0; // update buffered time
		}

		t0 += step;
	}
	return 0;
}
```

This is an issue which is not as easy to detect but still can cause serious problems, especially for time conditioned control loops.
To prevent such issues, ROS for instance tracks time using a uint32 for seconds and a separate uint32 for nanoseconds.

saran-t, 2022-10-24:
Agreed in principle but changing `time` to a pair of integers will likely break a lot of dependent code. What do you think about just introducing an additional uint64 step counter?

DavidPL1, 2022-10-25:
I think adding a step counter won't solve the problem, because one would have to multiply it with `timestep` to get the current time, causing the same issue. Besides, if `timestep` changes there is no way to resolve the current time based on the step counter alone.

I get that you want to keep the API stable to prevent dependent code from breaking, so instead of replacing `time`, would it be possible to add some sort of `precise_time` class/struct as member to `mjData`?
This class or struct could track seconds and milliseconds and define comparison operators.
Then, if precision is required, one would be able to directly use `mjData.precise_time` for arithmetic operations and otherwise `mjData.time` would still be available.

saran-t, 2022-10-25:
I disagree with your first point: multiplying a step counter by `timestep` will not accumulate errors in the same way as the `mjtNum` clock. You can quantify the precision of the multiplied result.

Point taken about changing `timestep`, but this isn't something that MuJoCo internally does (at least not today). In any case, since we'll not want to change `timestep` away from `mjtNum` (again, API compatibility), I don't see a clean way to support arbitrarily changing `timestep` without losing precision.

Can you please give us a concrete case where the time accumulation error is actually causing problems?

DavidPL1, 2022-10-26:
Well yes, you would have quantifiable precision with the uint64 mjtNum product. Assuming `timestep` remains fixed this solution works. I myself wasn't changing `timestep` in my code so far, I was just thinking about cases where this might break.
But even if MuJoCo is not changing `timestep` it is still possible and it is not discouraged, so this does not sound like a final solution.

Sidenote: Is backward stepping still supported and are negative times possible (I noticed that the backward stepping shortcut was removed in `simulate.cc`)? A uint64 step counter would certainly have problems with that.

The use cases that made me aware of this issue are streaming sensor data at a fixed rate and running a robot controller at a fixed rate. Once in a while you miss your rate by `timestep`. The severity of the problems directly correlates with the magnitude of `timestep` and is much more critical if MuJoCo is compiled to use floats instead of doubles for mjtNum. This is a problem that can be fixed e.g. by rounding, but the user has to be aware of this problem and manually find a workaround and that's where I see the issue.

In my opinion changing the way time is tracked is something that should be kept in mind for a major version update.

closed 2024-01-03 by kevinzakka