google-deepmind/mujoco #548: RK4 issue with Large Muscle Actuator , but Euler, Implicit work fine

MyelinsheathXD, 2022-10-26:
Hey after creating humanoid with large muslce force about 300 unit for every muscle
Humanoid body flying up after using its muscles.

This case occurs only when Integrator is RK4
whereas Euler and semi-implicit are fine.
I bealive this is common issue with  all physics engines , I have seen this similar effect with phyX 4.1's articulated body in Unity  engine.

Expected Solution: RK4 integrator should step the acticulated body properly when large actuator forces are applied  

I demonstrated this case in the video below.
[sample video](https://youtu.be/i-yvFz2SC5M)

- Include the following context:
  - Operating system== Win10
  - MuJoCo version == 2.2.2
  
[myfileHelloS.zip](https://github.com/deepmind/mujoco/files/9873557/myfileHelloS.zip)

quagla, 2022-10-27:
Hi, both Euler and semi-implicit actually treat damping implicitly. Are they stable if you deactivate damping? And is RK4 stable if you decrease the time step?

yuvaltassa, 2022-10-27:
This is to be expected. RK4 is not always the "best" integrator.
RK4 is exceptionally good at integrating _energy conserving_ systems, but for systems with large energy dissipation, especially dissipation due to various forms of damping, Euler and Implicit are usually better.

As @quagla says, if reducing the timestep makes the RK4 instability go away then there is no bug, just different properties of integrators.

closed 2022-10-27 by yuvaltassa