google-deepmind/mujoco #550: cable.dll plugin not registering

mathl79, 2022-10-27:
Hi, 

I am an engineer and want to use Mujoco for some process simulation. I am curious to use new elastic cable feature.

My problem is, that the cable plugin is not loaded by simulate.exe after successful compilation (Visual Studio 17 2022).

Here is, what i did so far:
First i created the solution files with the command `cmake -G "Visual Studio 17 2022"`

The compile process completed successfully and simulate.exe can be started.
I did copy the cable.dll into the \plugin\ subfolder located in the folder of the executable.

For some reason, the compiled cable.dll is not loaded as plugin like the precompiled cable.dll from the github releases.
I did not alter the source of it, but the filesize of the compiled version is 14kB instead of the precompiled file having a size of 33kB.

Using the cable.dll from the precompiled 2.3.0 release, the plugin is loaded and everything is working as expected.

What do i miss, to get the cable.dll recognized as plugin correctly ?

thanks in advance!

quagla, 2022-10-27:
Not an expert in this area, but probably @saran-t can help. I assume you are on the standard MSVC compilers/toolset?

saran-t, 2022-10-27:
Can you please try deleting `static` from https://github.com/deepmind/mujoco/blob/main/include/mujoco/mjplugin.h#L78 and see what happens?

mathl79, 2022-10-27:
removing `static` from line 78 in mjplugin.h does not solve the plugin detection issue.
[image]

mathl79, 2022-10-28:
with MSVC did not find the `int __stdcall DllMain()` entrypoint [https://github.com/deepmind/mujoco/blob/main/plugin/elasticity/cable.cc#L224](url) to be ever reached.

Maybe this has something to to with the different dll-initialization sequences ?
[https://stackoverflow.com/questions/22763945/dll-main-on-windows-vs-attribute-constructor-entry-points-on-linux](url)

I switched now to LLVM/Clang and the plugin is loaded from `simulate.exe` as expected.

saran-t, 2022-10-28:
Can you try adding `__declspec(dllexport)` where `static` used to be?

mathl79, 2022-10-28:
`simulate.exe` does still not load the dll with this new directive at mjplugin.h#L78. 
I tried debug and release configs.

quagla, 2022-10-28:
It would be nice to resolve this but if you need a quick and dirty hack to start developing you can always change https://github.com/deepmind/mujoco/blob/main/plugin/elasticity/CMakeLists.txt to add the `MUJOCO_ELASTICITY_SRCS` directly to the `mujoco` library instead of creating a separate DLL. You didn't hear it from me :)

mathl79, 2022-10-28:
i have found the solution!  it is caused by the Name-Mangling of DLL Exportnames.
we need to change that line [(https://github.com/deepmind/mujoco/blob/main/include/mujoco/mjplugin.h#L78](url) to
`extern "C" __declspec(dllexport) int DllMain(void* hinst, unsigned long reason, void* reserved) { \` to avoid the export name to be changed. 

credits to Jonas OSDevers Post: [https://www.c-plusplus.net/forum/topic/327317/dll-s-dynamisch-verwenden-funktioniert-nicht/4](url) (german)

mathl79, 2022-10-28:
`extern "C" int __stdcall mjDLLMAIN(void* hinst, unsigned long reason, void* reserved)` works also.

commit c7960385 (2022-10-28): Declare DllMain as `extern "C"`.

Fixes #550.

PiperOrigin-RevId: 484540205
Change-Id: I3915f65f1ae6ab3a93118b9d747b5c09d9abe872
closed 2022-10-28 by copybara-service[bot]