google-deepmind/mujoco #552: Naming collision friendly executable and plugin names?

traversaro, 2022-10-27:
Hello mujoco devs! For the packaging of the 2.3.0 in conda-forge (https://github.com/conda-forge/mujoco-feedstock/pull/10) I was looking into the new plugin support. I noticed that the only plugin shipped with mujoco is installed (not in the CMake sense, but looking at the released binaries) at `<install_prefix>/bin/plugin/cable.so` (or `.dll` on Windows).

As this path does not contain anything mujoco specific, when mujoco is installed in a prefix with other packages (as it happens when a package is installed by most package managers) this could lead to naming collisions. I never tought about this (as they are not currently packaged by the conda-forge package), but a similar point holds also for the executables shipped under `bin`, such as `bin/simulate`, `bin/record`, `bin/build`. Indeed, the recent debian packaging of mujoco renames them with the `mujoco-` prefix, see https://packages.debian.org/bookworm/amd64/libmujoco-samples/filelist . However, I think that `bin/plugin/cable.so` is more problematic as it is required for the full functionalities of the python bindings.

As mujoco is starting packaged by many package managers (see https://repology.org/project/mujoco/versions) I wonder if you may be interested in changing `bin/plugin/cable.so` to a more "naming collision friendly" path (such as `bin/mujoco-plugin/cable.so` or `bin/plugin/mujoco-cable.so`), or at least have a CMake option that permits to do that?

saran-t, 2022-10-28:
Yes, this actually occurred to me the day before we pushed out the release but it was a bit late to change it then.

`bin/mujoco-plugin` is good. I'm still not 100% certain whether these plugins should be treated as `bin` or `lib`, but I think `bin` makes sense since these are something that are more likely to get lazily loaded by applications.

Thoughts?

traversaro, 2022-10-28:
> `bin/mujoco-plugin` is good. I'm still not 100% certain whether these plugins should be treated as `bin` or `lib`, but I think `bin` makes sense since these are something that are more likely to get lazily loaded by applications.
> 
> Thoughts?

Initially I checked a few libraries for where they install their plugins, and apparently there is no consensus:
* Qt uses `<install_prefix>/plugins` (see https://doc.qt.io/qt-6/plugins-howto.html#locating-plugins)
* Modern Gazebo (i.e. formerly Ignition) uses `<install_prefix>/lib/<library_name>/plugins`

However, I then checked a random Ubuntu 22.04 that I have under my control, and I found a lot of subdirectories under `/usr/lib` (each one with the name of a library or program), and only one (`X11`) under `/usr/bin`. Indeed, by checking a bit more I found that:
* The FHS suggests in `/usr/lib` section suggests that "Applications may use a single subdirectory under /usr/lib." (see https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s06.html)
* Even if qt by defaults install plugins in `<install_prefix>/plugins`, the Debian packaging moves them in `/usr/lib/qt5` 

Given all of this, I guess `<install_prefix>/lib/mujoco` or `<install_prefix>/lib/mujoco/plugins` may be the option more in line with UNIX tradition, even if personally I do not feel strongly about it (I do not think that the FHS police will look for you if you put plugins in `bin/mujoco` :D ).

saran-t, 2022-11-15:
How about we go with `bin/mujoco_plugins` for the time being? I can finalise this for the upcoming 2.3.1 release.

Beyond 2.3.1, perhaps it would be a good idea for the plugin search path for the `simulate` binary to be configurable at build time. When we ship our prebuilt packages it's convenient to just stick plugin DSOs in a directory alongside `simulate` since that makes sense on all platforms (Windows doesn't have the equivalent of `lib/` for example). However, I agree that when "installed" into a Unix-compliant system, we should probably put plugins under `lib`.

traversaro, 2022-11-15:
> How about we go with `bin/mujoco_plugin` for the time being?

Good for me!

saran-t, 2022-11-16:
Fixed in 2ebc5f097b1a5fbbe276250ad1d13966510d1291

closed 2022-11-16 by saran-t