google-deepmind/mujoco #553: Is it possible to write a for loop to add bodies in the MJCF XML file ?

joaopfg, 2022-10-28:
Hi, suppose I want to generate `n` consecutive boxes in the scene. Is there some mechanism to write a for loop in the MJCF XML file ?

yuvaltassa, 2022-10-28:
The [composite](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-composite) tag does something like this, in particular look at the `particle` composite type.

Unfortunately I think it is quite limited as it currently only gives you 3 slider joints. @quagla, is there a way to make that free joints, in the current implementation?

closed 2022-11-07 by copybara-service[bot]

yuvaltassa, 2022-11-07:
So the commit solved this for adding an array of boxes, but we are thinking about a more general `<replicate>` element that will allow you to replicate anything along a grid (e.g. make multiple copies of robots). Will update here when that happens.

joaopfg, 2022-11-07:
@yuvaltassa Thanks ! It would be good if the `<replicate>` also gives the possibility of not being limited by a grid. For example, if I want to add many robots around a circular conveyor belt. I am thinking in something similar to the `loop` element ones that some web frameworks use. For example, Angular's [ngFor](https://blog.angular-university.io/angular-2-ngfor/) .

yuvaltassa, 2022-11-07:
What would you say to a choice of either grid of you can provide a mesh and <replicate> will use the vertices?

yuvaltassa, 2022-11-07:
Now that I write it I see that even if we use the normals for "z-orientation of thing", we have an indeterminism w.r.t the xy axes. So maybe mesh to not the most general thing. But maybe this is not so bad?

joaopfg, 2022-11-07:
@yuvaltassa I think just giving the possibility of parameterizing the loop is already very powerful.  For example, I could represent `n` robots around a circle of radius `r` in the xy plane with the following syntax:

```
<for begin="0" end="2*pi", step="(2*pi)/n">
    <body pos="r*cos(theta)  r*sin(theta)  0" quat="...">...<body/>
<for/>
```

What do you think ?

yuvaltassa, 2022-11-07:
I think it would be easier to provide a Python code sample that makes this shape as a mesh 🙂

Basically if we're going to do something more general anyway, I'd rather avoid MJCF schema inflation if I can.

joaopfg, 2022-11-07:
@yuvaltassa I see your point about MJCF schema inflation. Indeed, makes sense. I would be happy with meshes. I think a natural extension could be a set of unitary vectors and their corresponding origin points. Like this, one could fill an arbitrary volume with bodies in arbitrary orientations.

Niels-Frederik, 2023-02-14:
I really like the idea of a `<replicate>` tag, even with just the option of a grid. Is this still in the pipeline?

yuvaltassa, 2023-02-15:
@Niels-Frederik if you're happy with Python, you can use dm_control's [pyMJCF](https://github.com/deepmind/dm_control/blob/main/dm_control/mjcf/README.md).

Niels-Frederik, 2023-02-19:
I have actually just switched to C++ due to the simulated environment being rather large, and performance being quite critical. Is the best approach then to just simply generate the XML string of the model dynamically through c++?

commit 3b89b0fd (2022-11-07): Particles with arbitrary geoms and joints.

Fixes #553

PiperOrigin-RevId: 486614769
Change-Id: I9a09bf73da97b6b9746800a72ac6c97cce2efa92
commit 26f23066 (2024-04-30): Add `replicate` element to MJCF. Fixes #553

PiperOrigin-RevId: 629393578
Change-Id: Ia1db1b9e47eba8868117b0f7b85edcd0cc1b6d79