google-deepmind/mujoco #558: Index of joint-space inertia matrix (JSIM)

zxwang96, 2022-11-03:
Hi,

I'm building a controller to control a UR5 robot to manipulate a rope object (96 linked object). When I use the `mj_fullM` function to convert `data.qM` to the dense format, I found it's dimension is 196 x 196. To control the UR5, I only need the JSIM of the 6 joints of UR5. My question is, how can I index the 196 x 196 JSIM to get the JSIM of UR5? I attached my zipped MJCF file below. Thanks.

[ur5_rope_c.zip](https://github.com/deepmind/mujoco/files/9926677/ur5_rope_c.zip)

yuvaltassa, 2022-11-06:
Are you using Python or C? The answer is different depending on how you are accessing MuJoCo.

zxwang96, 2022-11-07:
Hi @yuvaltassa, I'm using Python.

yuvaltassa, 2022-11-07:
Firstly, the order of joints is the same as in the XML. So maybe you just know the indices by virtue of where the robot is defined. If you're not sure you can do
```py
joint_indices = [model.joint(i).id for i in range(model.njnt) if `UR5` in model.joint(i).name]
```
Or whatever substring will identify your robot joints. You can also give all your robot joints some user value
```xml
<joint ... user="1"/>
```
and do 
```py
joint_indices = [model.joint(i).id for i in range(model.njnt) if model.joint(i).user[0] == 1]
```
Or similar.

Thenm after calling `mj_fullM` you just slice:
```py
M[joint_indices, joint_indices]
```

closed 2022-11-07 by yuvaltassa