google-deepmind/mujoco #559: How would you go about normalizing MuJoCo observations for a neural network?

MotorCityCobra, 2022-11-03:
Sorry if this question is outside the scope of this repo, but I think if anyone would know how to normalize this data for training if might be someone here.

This is my code taking observations, just to give an idea of what I'm observing. It's 385 numbers.  

```
    torch::Tensor qpos_T = torch::from_blob(d[id]->qpos, {28}, options);
    torch::Tensor qvel_T = torch::from_blob(d[id]->qvel, {28}, options);
    torch::Tensor cinert_T = torch::from_blob(d[id]->cinert, {140}, options);
    torch::Tensor cvel_T = torch::from_blob(d[id]->cvel, {84}, options);
    torch::Tensor qfrc_actuator_T = torch::from_blob(d[id]->qfrc_actuator, {21}, options);
    torch::Tensor cfrc_ext_T = torch::from_blob(d[id]->cfrc_ext, {84}, options);
```

This is a small sample of one observation. Some numbers are near zero. Some are 20 and 30 decimal places to the right or left.  

```
 1.2436e-27 -2.4532e+00  1.5884e+18 -2.2668e+00 -2.6081e-11  1.4253e+00
 4.8692e-14 -2.1157e+00  1.7720e-22 -1.9623e+00  1.0951e+30  1.5578e+00
 7.0858e-30 -2.2597e+00  1.2190e+22  2.1017e+00  1.1557e+36  1.2906e+00
 3.5779e-03  2.0355e+00  3.7871e+07  2.1378e+00 -1.0378e-07  1.9220e+00
-7.4832e-06 -1.5724e+00 -1.5107e-02 -1.9287e+00 -1.4866e-26  1.6272e+00
 3.1657e-14 -1.9924e+00 -9.4101e-37 -2.0469e+00 -4.1626e-01 -1.3577e+00
 5.4301e-29 -1.6695e+00 -1.8124e-25  2.5714e+00  3.6762e+17  1.6995e+00
-2.6828e-29  2.3965e+00 -3.9656e+19  2.4138e+00  1.9539e-03  1.8795e+00
 2.4330e-08  1.7665e+00 -7.5912e+10 -1.7527e+00  3.8436e+25 -1.7918e+00
 3.8472e+02  2.4023e+00  1.8100e+28 -1.6236e+00  2.3529e-16  1.5328e+00
 3.1171e+07  1.7642e+00 -3.1588e-18  1.6832e+00 -4.6191e-20  1.9147e+00
 2.5456e-28 -2.0834e+00 -2.2753e+01  2.3255e+00  3.9845e+28  1.7157e+00
 5.0635e+32  2.4273e+00 -7.0333e+27 -1.2354e+00  1.7251e-18  1.6025e+00
 3.6919e-22  1.9400e+00  2.3794e-33 -2.3246e+00 -1.2525e-05  1.8876e+00
-2.6047e+20 -1.5295e+00 -6.1313e-18 -1.8871e+00 -6.5229e-32 -1.6041e+00
-1.4765e-20 -2.1831e+00 -3.7710e+34 -2.2575e+00 -2.3642e-25  1.7687e+00
 2.5841e+01 -2.0837e+00 -1.3663e-09 -1.7502e+00  3.8335e+31 -1.7498e+00
 1.7964e-25 -2.3413e+00 -7.4443e-39  2.2477e+00 -1.9026e-04 -1.5578e+00
 8.7306e-17  2.4791e+00 -1.9048e-28 -1.9820e+00  8.6059e-20  2.0325e+00
 2.7356e-11 -2.1882e+00  7.4750e-11 -1.8847e+00  3.5613e+00 -1.7681e+00
 2.7969e-01 -2.0107e+00  3.1666e+09  1.8748e+00 -1.0512e-25  1.7832e+00
 1.0904e+36 -1.5607e+00 -5.3314e+18  1.6612e+00 -1.7101e-04 -1.6972e+00
-7.8255e+10  2.0960e+00 -3.5076e-02  2.0395e+00  1.3489e-02 -1.6592e+00
-9.2899e+20  2.1187e+00  1.2068e+01  2.6163e+00  1.8799e-37 -1.9909e+00
-2.1188e+25  2.3268e+00 -9.5844e+05  1.5513e+00 -4.2476e-17  1.7992e+00
 1.4408e-28  2.0031e+00 -5.6622e+11 -2.0779e+00  8.0727e+08 -1.5190e+00
 4.2466e-26 -2.0074e+00  1.0121e+14  1.8832e+00 -1.5463e-11 -1.6730e+00
```

Balint-H, 2022-11-04:
If you have a nominal reference data you could calculate normalization parameters from that for each observation. For joint positions, you could scale inputs each step based on their joint limits for example. 

Alternatively you could implement and calculate standard scaled array of observations, updated each step. Or depending on your batch size, you might be able to get away with just adding a batch norm at the start?

MotorCityCobra, 2022-11-04:
@Balint-H Thank you.  

I'm not sure about batch norm "at the start" in LibTorch. This is one of the two models. The actor model.  
I've been thinking that I should try batch norm and max pooling. I'm honestly not sure what they are, what they do and how to implement in LibTorch.  
```

struct Act_Net : torch::nn::Module {
    torch::Tensor action_high, action_low;
public:
    Act_Net(torch::Tensor action_high, torch::Tensor action_low)  : action_high(action_high), action_low(action_low){
        // Construct and register two Linear submodules.
        fc1 = register_module("fc1", torch::nn::Linear(385, 385));
        fc2 = register_module("fc2", torch::nn::Linear(385, 42));
        fc3 = register_module("fc3", torch::nn::Linear(42, 42));
        // cholesky_layer = torch::nn::Linear(385, (42 * (42 + 1)) / 2);
        cholesky_layer = register_module("cholesky_layer", torch::nn::Linear(42, 42));
        fc1->to(device);
        fc2->to(device);
        fc3->to(device);
        cholesky_layer->to(device);
    }
    // Implement the Act_Net's algorithm.
    torch::Tensor forward(torch::Tensor x) {
        // Use one of many tensor manipulation functions.
        x = torch::relu(fc1->forward(x));
        // x = torch::dropout(x, /*p=*/0.2, /*train=*/is_training());
        x = torch::tanh(fc2->forward(x));
        auto mean_layer = torch::relu(fc3->forward(x));
        mean_layer = torch::nan_to_num(mean_layer, 0);
        auto mean = action_low + (action_high - action_low) * mean_layer;

        auto chol_l = cholesky_layer->forward(x);

        auto chol = torch::matmul(chol_l, chol_l.transpose(0, 1));
        chol = torch::nan_to_num(chol, 0, 2.0);
        chol = chol.add(torch::eye(42).to(device));
        auto cholesky = torch::linalg::cholesky(chol);

        return torch::cat({mean, cholesky}, 1);
    }

    torch::Tensor action(torch::Tensor state) {
        torch::NoGradGuard no_grad;
        auto forward_output = forward(state);
        auto forward_mean = forward_output.index_select(1, torch::arange(42).to(device));
        auto forward_cholesky = forward_output.index_select(1, torch::arange(42, 84).to(device));
        auto forward_action = MultivariateNormal_sampler(1, forward_mean.to(device), forward_cholesky);

        return forward_action.squeeze(0);

        // auto batch_actions = forward_action.index_select(1, torch::arange(42));
        // auto one_action = batch_actions.index({0});
        // return torch::tanh(one_action);
    }

    // Use one of many "standard library" modules.
    torch::nn::Linear fc1{nullptr}, fc2{nullptr}, fc3{nullptr}, cholesky_layer{nullptr};
};
```

Balint-H, 2022-11-04:
You would only use maxpool if the structure/order of the units in your layer holds meaningful information about redundancy, and you want to take advantage of that and reduce your dimensionality. In most control situations you wouldnt benefit from them, unless e.g. you use a spatial matrix of IMU sensors, use visual inputs or use a temporal window of observations, in which case you can use it in the context of temporal convolutional networks. Batch norm layers effectively take all the data you provided the network in one round of learning, and normalizes the signals for each unit separately.

MotorCityCobra, 2022-11-05:
@Balint-H  I really appreciate the spoon feeding. It is much more helpful for me to learn what these things are here now in context rather than look up what they generally do.

closed 2022-11-05 by MotorCityCobra