google-deepmind/mujoco #564: stl model cannot import

turnnext, 2022-11-07:
I'm a studentand I'm trying to use MuJoCo for  my subject.

Here is a model which explains my question:

I have no problem with the xml code, but my STL model import always doesn't work, the STL file is binary, and I also tried to shrink the STL file size

kevinzakka, 2022-11-08:
Can you attach your STL? And putting the error message would be helpful (bottom left corner of simulate app).

kevinzakka, 2022-11-16:
To clarify, MuJoCo failing to load an STL file is usually because that STL file is too big (in which case you probably want to downsample it). In that case, the error message should be pretty descriptive. If you are running into another issue, attaching it so that I can debug would be great :)

kevinzakka, 2022-12-06:
Closing for now, feel free to re-open if you are still encoutering an issue with the STL file.

closed 2022-12-06 by kevinzakka